universam1/iSpindel #468: Battery only lasting 6-7 days

Haggis2020, 2021-02-19:
Can someone help fix a battery problem I have. I just received an assembled ispindel with PCB 4.0 board.
Everything is working fine except the battery is draining very quickly. The first time I used it it dropped from 4v to 3.25v in just over 6 days then stopped sending data. I assume because the battery voltage was to low.
The ispindel is set to send data to ubidots every 900s and while testing I see the the blue LED on the WEMOS blinks every 15 minutes
I have flashed the latest firmware (6.5.1)
I have tried 2 different 18650s. Samsung 25R supplied with the ispindel and a Molicel P26A bought from Fogstar.

Does anyone have any ideas why the battery is draining so quickly.

pppedrillo, 2021-02-20:
Check with iSpindel off. If battery drains when it off, you either have a problem with soldering or current leakage via a dodgy charging module.

Haggis2020, 2021-02-20:
I have had the ispindel switched off for 3 1/2 hours and there has been no voltage drop.

pppedrillo, 2021-02-20:
It might drain slower then off. I'd let it couple of days before re-check.

thegreatgunbantoad, 2021-02-20:
@Haggis2020 most likely it's build with a Wemos clone ESP board.  Some of those boards are just awful.  If you can, I'd suggest you replace that board.  I have heard of this issue a few times and I do have a unit that does this somewhere.  I'll try to track it down an get around to diagnosing the fault properly.

Haggis2020, 2021-02-21:
I don't know if this helps but here is the serial monitor

Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel000","Token":"BBFF-bS85DST3yqpTKlAZo6yKiCvEbC6OcZ","Sleep":3600,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"XXXXXXXXXXX","PSK":"XXXXXXXXXXXX","Offset":[0,0,0,0,0,0]}
offsets not available
Boot-Mode: External System
power-cycle or reset detected, config mode

going to Config Mode
*WM: Scan done
*WM: DUP AP: Virgin Media
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started

ErikdBr, 2021-02-21:
I see "Parse error at 1" between temp and gravity just above the calling ubidots phrase. Also when a reset is detected I think it is not normal it leaves the config mode by itself after waiting 2.38 seconds. Also a gravity of 0.
I think you have a shortage somewhere.

pppedrillo, 2021-02-21:
@ErikdBr "Parse error at 1" is an error in poly parsing - sort of typo there or such, hence gravity == 0

Haggis2020, 2021-02-22:
The battery voltage stays the same if the ispindel is off but drops about 0.01v per hour when sending data every 900s or 3600s. Is there anything else I could look at. I could return it but as I'm sheltering at present this going to be difficult which is why I'm trying to solve the problem.

closed 2021-05-01 by stale[bot]