universam1/iSpindel #479: Extract ispindel raw data

nkaramolegos, 2021-03-17:
Hello, I was wondering if I can get the temperature via an API in order to get the temperature in my python program. Specifically, I would like to build a http (or MQTT) server in python to watch the temperature inside my kegs with the ispindel.

pppedrillo, 2021-03-20:
Can't you simply parse json and get a temp, pretty much as the other sws do?
Also the idea sounds a bit "overengineered" to me. If your kegs are in insulated chamber with a temp  control (ie kegerator/keezer/fridge), after short transition period temperature inside the keg will be exactly the same as outside the keg, isnt it?

nkaramolegos, 2021-03-20:
Maybe is over engineering but DIY is fan. So the JSON is the responce of the ispindel to the server? What is the format?

pppedrillo, 2021-03-20:
> 
> 
>So the JSON is the responce of the ispindel to the server? What is the format?

https://github.com/universam1/iSpindel/blob/934cf2e9e7060ded7cffa17e9b9211da4fda2658/pio/src/iSpindel.cpp#L544

closed 2021-05-28 by stale[bot]