universam1/iSpindel #483: Issue with previously working and flashed ISpindel

pietro-22, 2021-03-20:
Hello, I have a question related to flashing the d1 mini. I have built two ISpindels, both with the pcb from opensourcedistilling. Previously, both of my units were working/have been flashed, and I was able to connect to them via wifi. Original problem was with the gyroscope, receiving NaN values, even with the most recent firmware. I opted to swap out the gyroscopes and as luck would have it, only one of the ISpindels will connect to wifi. My problem is in trying to flash the non working unit. When I power on the ISpindel, I see a solid blue light. After locating the firmware and pressing flash, the blue led blinks once and then turns off. It seems like the flash process is stalling. I can see in the nodemcu log that it says "Begin find ESP8266". I have also tried connecting it to Putty to see if I could find the problem, but it is just showing constantly refreshing gibberish. Battery is fully charged. Would some problem with the gyroscope completely throw off the boot process of the d1 mini? please help, I have been trying to build these for a few months now and have had individual problems with individual parts at each turn.

thegreatgunbantoad, 2021-03-28:
Solid blue light on power up usually means knickered or disconnected temp sensor.
Also bear in mind there are a lot of duff ESP boards out there.

bolek193, 2021-03-30:
> Hello, I have a question related to flashing the d1 mini. I have built two ISpindels, both with the pcb from opensourcedistilling. Previously, both of my units were working/have been flashed, and I [...]

closed 2021-06-06 by stale[bot]