universam1/iSpindel #489: wemos config corrupted

Wayneski69, 2021-03-28:
Hi,
I have build my ispindel but have issues when flashing the wemos. I can flash and connect to the config pages but cant save the config. I have tried using NodeMCU PyFlasher and the NODEMCU flasher as recommended. I have also tried wiping the data as recommended in various reports but still no joy. I have pasted the log when resetting below. I have managed to connect to UBIDOTS once and had one reading but then no more, also I cant figure out how to paste the full serial log into this post. Thanks for any help.

FW 6.5.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... ERROR: failed to mount FS!

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 E5 55 76 E0 1 3C 24
  Chip = DS18B20
  Data = 1 57 1 55 5 7F A5 81 66 45  CRC=45
  Temperature = 21.44 Celsius, 
offsets not available
Boot-Mode: External System
power-cycle or reset detected, config mode
...........
ERROR no Wifi credentials

going to Config Mode
reboot RFCAL
rl

ErikdBr, 2021-03-29:
'offsets not available' could indicate a dodgy MPU-6050.

bolek193, 2021-03-30:
> MPU

Wayneski69, 2021-04-03:
Changed the MPU and the Wemos for good measure and all seems well now. Thank you for your help.

closed 2021-06-05 by universam1