universam1/iSpindel #496: Make even nicer for battery life -- idea list

thormj, 2021-04-04:
## I'm submitting a ...
  - [ ] bug report
  - [X] feature request
  - [ ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Context
Ideas to help us conserve battery power:

**1.**  [Pull request #485] If placed in a certain orientation (like the commercial verson) sleep harder/longer  -- this needs to "not be normal during fermentation" so you don't go to sleep while you're brewing (battery up or on cap or....).
2. Since we only use the accelerometer, only use that part of the chip (500uA for accelerometer vs 3.9mA for everything)
2. Instead of waking up every so often, use the accelerometer's interrupt to instant-wake (may require wiring) -- would make the Tilt Commands even more efficient since we wouldn't interrogate the MPU-6000A every 30 sec.
3. Add the concept of debug levels so that there is little/no serial output if we're not hooked to a PC (we're not wasting much power on the serial port, but we are "hanging out" while we're printing the messages at 115200 baud).
4. Use the temperature sensor of the MPU-6000A instead of the DS1820[B] (how does that reflect the wort temp?)
5. I kinda like the blinky blue LED, but... add measurements/instructions for removing it for advanced users
6. A battery-schedule (if horizontal [early ferm], every 10 minutes, if near vertical [end game], every 8 hours]... or something like that.

Any more ideas for power saving?
I figure we could take a number and make a new issue out of it and I'll add a * by the ones that have been looked at
--
Resources: 
  Datasheet: http://cdn.sparkfun.com/datasheets/Components/General%20IC/PS-MPU-6000A.pdf
  Registers: http://cdn.sparkfun.com/datasheets/Sensors/Accelerometers/RM-MPU-6000A.pdf'

mp-se, 2021-04-06:
I've been doing some testing on point 4. I've had a device that is in sleep mode for 30s and at startup measure the temperature from the accelometer and DS18B20 sensor. The difference is on my device +/- 0.3 C over a 2 hour test window. I will do some more tests during a fermentation to compare the two. If the device is powerd on for a long time the differnence will be up to +4C which is not really strange.

ErikdBr, 2021-04-06:
> 4. Use the temperature sensor of the MPU-6000A instead of the DS1820[B] (how does that reflect the wort temp?)

The DS18B20 is in all PCB designs as close to the wort as it ever can get and for sure closer to the wort as the one from the MPU-6050 (The Jeffrey 2.69 PCB, the @mikmonken V2.1 PCB, the @Gromitdj V2.0 PCB and the @cherryphilip 4.0 PCB amongst others)

> 5. I kinda like the blinky blue LED

Me too, and the green/red one on the MPU-6050 also. I don't remove anything, no leds and no diode and get a battery life of 80 days on a report interval of every 15 minutes. I guess when I set the interval to every half hour I will get a battery life of 100+ days.

So why do you want an even better battery life?

> 6. A battery-schedule (if horizontal [early ferm], every 10 minutes, if near vertical [end game], every 8 hours]... or something like that.

I want my iSpindel report at the interval I set no matter which part of fermentation. As long as I don't switch it off, It should keep reporting wether it's standing up straight, laying down at 90 degrees or whatever angle you can think of.

m0n5t3r, 2021-05-30:
not sure how offtopic this is, but throwing it in anyway: I have some small changes in my fork to allow for LiFePO4 cells - does a PR make sense? https://github.com/universam1/iSpindel/compare/master...m0n5t3r:master (own design, 30mm tube with 700 mAh 14500 cell that lasts about 2 months; I haven't published the board yet because I want to foolproof it a little - it needs reverse polarity protection and a voltage supervisor, since I've had batteries go flat on some long fermentations - let's say I put too much honey in my last mead and I kept waiting to see if the flatline is for good - and maybe try the trick with a big solder pad for balancing)

closed 2021-08-05 by stale[bot]