universam1/iSpindel #498: iSpindel does not wake after first submission

Timswolgen123, 2021-04-21:
## I'm submitting a ...
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x] Checked the logs for errors in the terminal console ?
- [x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?
- [x] Looked through similar reports already and found that it could have something to do with a BAT43 diode. I checked this, and mine is placed in the right direction. 
- [x] Flashed the device several times. This does not seem to work.
- [x] Measured the voltage over the D0 and RST pin. When the device is in sleep mode, the voltage is 0V. When the device is reset, so when the data is being submitted, the voltage has a spike to about 1V. When the led turns on for a moment after 10 seconds (after the first interval), it has a really small spike of about 0.02V. When the led turns on when the first interval is finished, the led only turns on half a second. Obviously way shorter than when the reset button is pressed and the data is submitted.

## Report

The iSpindel only logs the first data interval so when the device is switched on or when the reset button is pressed. After this first data submission, the  device does not wake anymore. I've found several similar issues in the past but they all seem to be from a few years ago. I tried to go through every troubleshooting step I could find but did not find the solution to my issue. I am starting to think I could just better buy a new Wemos D1 mini. 

### Console Logs

This is the serial monitor after the device is switched on or when the reset button is pressed. Nothing major to see here. (With the knowledge I've gained from the all the other similar issues I found online.)
[image]

I've set the time interval to 10s. When the first interval is finished, these weird lines appear in the serial monitor. After this, nothing happens.
[image]

## Your Environment
- iSpindel The Jeffrey PCB v2.1
- Firmware Version used: 6.5.1
- Service backend you have configured the iSpindel to talk to: Brewfather

thegreatgunbantoad, 2021-04-21:
Check your diode connection.  It's usually that.

Timswolgen123, 2021-04-21:
> Check your diode connection. It's usually that.

What do you mean by connection exactly? The diode is placed the right way around. Do you suggest bad soldering?

thegreatgunbantoad, 2021-04-21:
Orientation of course.  But also check all the way to the pins on the Wemos.  Sometimes the clone Wemos boards have some pretty naff joints. (If you've bought from eBay or Ali you probably have a clone board).

Basically you're not waking from deep sleep correctly.

pppedrillo, 2021-04-22:
> 
> 
> Orientation of course. But also check all the way to the pins on the Wemos. Sometimes the clone Wemos boards have some pretty naff joints. (If you've bought from eBay or Ali you probably have a clone board).
> 
> Basically you're not waking from deep sleep correctly.
> Could also just be a duff Wemos/ESP board.

This is not a diode. This is SW crash.

ErikdBr, 2021-04-22:
If the D1 doesn't have the brandname 'LOLIN' on it you probably have a faulty clone.

Timswolgen123, 2021-04-22:
> If the D1 doesn't have the brandname 'LOLIN' on it you probably have a faulty clone.

Thankyou for the quick response!

I think that I indeed have a faulty clone. Could not find 'LOLIN' on the board. I also think I can confirm this because of the picture below. It states D1 m1n1 instead of mini. What are your thoughts on this?

[image]

ErikdBr, 2021-04-22:
Defenitly a clone, I don't say every clone is dodgy but I am done with clones, I allways had trouble with them. This is the original:
[image]
[image]

Timswolgen123, 2021-04-22:
> Defenitly a clone, I don't say every clone is dodgy but I am done with clones, I allways had trouble with them. This is the original:
> [image]
> [image]

Thanks for the help! I will order a new one.

closed 2021-04-22 by Timswolgen123

jurkstas, 2021-08-22:
I am having this issue with this same clone. Has anyone found the root cause?
When it is supposed to wake up, it sends out this at 74880 baud rate:
```
 ets Jan  8 2013,rst cause:5, boot mode:(3,6)

ets_main.c
```
[Here ](https://www.reddit.com/r/esp8266/comments/ke7s7z/what_the_heck_is_rst_cause5/gg32an9?utm_source=share&utm_medium=web2x&context=3) it is explained that cause:5 is unknown wake-up cause.

What is strange, that D0 is not going down at all. I also checked the power rails, LCR shows 14 uF capacitance and I don't see voltage dropping more than 50 mV during the entire cycle from reset to crashing/hanging.