universam1/iSpindel #507: No WIFI connection w/ FW 7.0.0

peterschrott, 2021-06-09:
## I'm submitting a ...
  - [x] bug report
  - [ ] feature request
  - [X] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [X] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [X] Checked the logs for errors in the terminal console ?
- [X] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

After updating the firmware of my iSpindel from 6.5.1 to 7.0.0 the d1 mini does not connect to my WIFI properly.

On the initial configuration of the wifi it seems to connect (see logs below).

On the wakups for reporting to ubidots somehow the connection is lost and it's restored. But the data are not sent do ubidots and on the next cycle it's restring the connection again.

I flashed the old 6.5.1 version again and wifi worked like charm. I also changed SSID & PW of my network -> no fix here.

### Console Logs

On saving WIFI preferences in web ui:
```
[...]
16:39:20.232 -> *WM: Sent wifi save page
16:39:22.230 -> *WM: Connecting to new AP
16:39:22.230 -> *WM: Connecting wifi with new parameters...
16:39:22.269 -> *WM: previous settings invalidated
16:39:22.447 -> *WM: After waiting...
16:39:25.651 -> *WM: 3.21
16:39:25.651 -> *WM: seconds
16:39:25.651 -> *WM: Connection result: 
16:39:25.651 -> *WM: WL_CONNECTED
16:39:25.651 -> saving config...
16:39:25.721 -> {"Name":"iSpindel000","Token":"xxx,"Sleep":15,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"ispindodal2","PSK":"xxx","Offset":[0,0,0,0,0,0]}
16:39:25.759 -> saved successfully
16:39:27.969 -> Samples:30 min:90.14 max:91.02 time:2193
16:39:27.969 -> x: -12288 y: 11172 z: -200
[...]
```

When waking up to report to ubidots:
```
[...]
16:44:04.230 -> FW 7.0.0
16:44:04.230 -> 2.2.2-dev(38a443e)
16:44:04.230 -> Worker run!
16:44:04.268 -> mounting FS... mounted!
16:44:04.268 -> reading config file
16:44:04.268 -> parsed config:
16:44:04.268 -> {"Name":"iSpindel000","Token":"xxx","Sleep":15,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"ispindodal2","PSK":"xxx","Offset":[0,0,0,0,0,0]}
16:44:04.447 -> offsets not available
16:44:04.447 -> Boot-Mode: Deep-Sleep Wake
16:44:04.447 -> 
16:44:04.447 -> woken from deepsleep, normal mode
16:44:06.656 -> Samples:30 min:90.19 max:91.01 time:2196
16:44:06.656 -> x: -12292 y: 11236 z: -196
16:44:06.656 -> Tilt: 90.66
16:44:06.656 -> Tacc: 42.84
16:44:06.656 -> Volt: 4.27
16:44:06.690 -> Temp: 26.50
16:44:06.690 -> Gravity: 33.89
16:44:06.690 -> After waiting 0ms, result 0
16:44:06.690 -> Failed to connect -> trying to restore connection...
16:44:06.690 -> Rescued Wifi credential
16:44:06.690 ->    -> waited for 3925ms, result 3
16:44:10.606 ->    -> Connection restored!
16:44:10.606 -> Final-sleep: 15s; RT: 6448
[...]
```

## Steps to Reproduce (for bugs)
1. Flashing iSpindel w/ FW 7.0.0
2. Setting up WIFI
3. Storing WIFI settings

## Your Environment
- Firmware Version used: 7.0.0
- Service backend you have configured the iSpindel to talk to: ubidots

pppedrillo, 2021-06-09:
> 16:44:06.690 -> After waiting 0ms, result 0

WL_IDLE_STATUS      = 0,

pppedrillo, 2021-06-09:
According to sources it will not send anything after " -> Connection restored!", just will go to sleep. Does it make any sense?

peterschrott, 2021-06-10:
@pppedrillo, can confirm, saw exactly this in the code and thought about changing the behavior such that after a successful "rescue" the data are sent. 
But then I had 2 concerns:
1) Why is it behaving like it does after fw upgrade? I did not see any suspicious in the the code changes from 6.5.1 -> 7.0.0. 
2) Not sure what it does to the battery, if the WIFI connection has to be "rescued" on every wakeup.

Changing the code in that manner would solve the issue of not uploading data. But not the issue of loosing wifi connection.

pppedrillo, 2021-06-10:
1. Its hard to say. There were some changes regarding set STA mode, may be those affects somehow? not sure
16:44:06.690 -> After waiting 0ms, result 0    means wifi is still in standby mode, not even tries to connect or do something useful 

2. I dont think it will change anything battery-wise. Sounds like your wifi is just still in standby at the time it wants to send a data, then it is waking up during that "rescue". 

What the wemos module do you have? can it be dodgy? I have 7.0 on 2 of mine ispindels and they send data to ubidots pretty well.

peterschrott, 2021-06-10:
1. I commented those changes out. This did not change anything in the behavior. 

2. I can change the logic here, such that data are sent out after success full "rescue". Maybe even open up an PR. But from a developers point of view, this solution is not quite satisfying to me as it's only a makeup for some other unknown issue.

I have ordered this d1 mini: https://www.reichelt.de/d1-mini-esp8266-v3-0-d1-mini-p253978.html. Haven't noticed anything wrong so far. As said, with fw v6.5.1 it all works just fine. Wifi seems awake and its sends out data right away.

Saving WIFI settings w/ 6.5.1:
```
[...]
12:16:13.868 -> *WM: Sent wifi save page
12:16:15.733 -> *WM: Connecting to new AP
12:16:15.733 -> *WM: Connecting wifi with new parameters...
12:16:15.733 -> *WM: previous settings invalidated
12:16:16.086 -> *WM: After waiting...
12:16:19.425 -> *WM: 3.35
12:16:19.425 -> *WM: seconds
12:16:19.425 -> *WM: Connection result: 
12:16:19.425 -> *WM: WL_CONNECTED
12:16:19.425 -> saving config...
12:16:19.486 -> {"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"___erster_low","PSK":"xxx","Offset":[0,0,0,0,0,0]}
12:16:19.486 -> saved successfully
12:16:20.045 -> Samples:30 min:89.19 max:90.24 time:518
12:16:20.045 -> x: -14852 y: 6884 z: 188
[...]
```

On wakeup w/ 6.5.1:
```
[...]
12:22:50.624 -> FW 6.5.1
12:22:50.624 -> 2.2.2-dev(38a443e)
12:22:50.624 -> Worker run!
12:22:50.624 -> mounting FS... mounted!
12:22:50.662 -> reading config file
12:22:50.662 -> parsed config:
12:22:50.662 -> {"Name":"iSpindel000","Token":"BBFF-2gObDxJcRstAGoqNfbNOorCvsNEbaj","Sleep":20,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"xxx","PSK":"17-Karotten","Offset":[0,0,0,0,0,0]}
12:22:50.847 -> Boot-Mode: Deep-Sleep Wake
12:22:50.847 -> 
12:22:50.847 -> woken from deepsleep, normal mode
12:22:51.618 -> Samples:42 min:89.24 max:90.13 time:748
12:22:51.618 -> x: -14868 y: 7020 z: 100
12:22:51.618 -> Tilt: 89.70
12:22:51.618 -> Tacc: 42.36
12:22:51.618 -> Volt: 4.26
12:22:51.618 -> Temp: 25.88
12:22:51.618 -> Gravity: 33.41
12:22:51.618 -> IP: 192.168.0.181
12:22:51.655 -> 
12:22:51.655 -> calling Ubidots
12:22:51.655 -> {"tilt":89.69512,"temperature":25.875,"battery":4.264859,"gravity":33.41217,"interval":20,"RSSI":-83}
12:22:51.836 -> Sender: Ubidots posting
12:22:51.836 -> POST /api/v1.6/devices/iSpindel000?token=xxx HTTP/1.1
12:22:51.836 -> Host: things.ubidots.com
12:22:51.836 -> User-Agent: ESP8266
12:22:51.836 -> Connection: close
12:22:51.836 -> Content-Type: application/json
12:22:51.836 -> Content-Length: 101
12:22:51.836 -> 
12:22:52.338 -> HTTP/1.1 200 OK
12:22:52.338 -> Server: nginx
12:22:52.338 -> Date: Thu, 10 Jun 2021 10:22:52 GMT
12:22:52.338 -> Content-Type: application/json
12:22:52.338 -> Transfer-Encoding: chunked
12:22:52.338 -> Connection: close
12:22:52.338 -> access-control-allow-origin: *
12:22:52.338 -> 
12:22:52.338 -> c0
12:22:52.338 -> {"battery":[{"status_code":201}],"gravity":[{"status_code":201}],"interval":[{"status_code":201}],"rssi":[{"status_code":201}],"temperature":[{"status_code":201}],"tilt":[{"status_code":201}]}
12:22:52.449 -> 0
12:22:52.449 -> 
12:22:52.449 -> 
12:22:52.449 -> Final-sleep: 20s; RT: 1903
[...]
```

peterschrott, 2021-06-10:
1. I commented those changes out. This did not change anything in the behavior. 

2. I can change the logic here, such that data are sent out after success full "rescue". Maybe even open up an PR. But from a developers point of view, this solution is not quite satisfying to me as it's only a makeup for some other unknown issue.

I have ordered this d1 mini: https://www.reichelt.de/d1-mini-esp8266-v3-0-d1-mini-p253978.html. Haven't noticed anything wrong so far. As said, with fw v6.5.1 it all works just fine. Wifi seems awake and its sends out data right away.

Saving WIFI settings w/ 6.5.1:
```
[...]
12:16:13.868 -> *WM: Sent wifi save page
12:16:15.733 -> *WM: Connecting to new AP
12:16:15.733 -> *WM: Connecting wifi with new parameters...
12:16:15.733 -> *WM: previous settings invalidated
12:16:16.086 -> *WM: After waiting...
12:16:19.425 -> *WM: 3.35
12:16:19.425 -> *WM: seconds
12:16:19.425 -> *WM: Connection result: 
12:16:19.425 -> *WM: WL_CONNECTED
12:16:19.425 -> saving config...
12:16:19.486 -> {"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"___erster_low","PSK":"xxx","Offset":[0,0,0,0,0,0]}
12:16:19.486 -> saved successfully
12:16:20.045 -> Samples:30 min:89.19 max:90.24 time:518
12:16:20.045 -> x: -14852 y: 6884 z: 188
[...]
```

On wakeup w/ 6.5.1:
```
[...]
12:22:50.624 -> FW 6.5.1
12:22:50.624 -> 2.2.2-dev(38a443e)
12:22:50.624 -> Worker run!
12:22:50.624 -> mounting FS... mounted!
12:22:50.662 -> reading config file
12:22:50.662 -> parsed config:
12:22:50.662 -> {"Name":"iSpindel000","Token":"BBFF-2gObDxJcRstAGoqNfbNOorCvsNEbaj","Sleep":20,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"___erster_low","PSK":"xxx","Offset":[0,0,0,0,0,0]}
12:22:50.847 -> Boot-Mode: Deep-Sleep Wake
12:22:50.847 -> 
12:22:50.847 -> woken from deepsleep, normal mode
12:22:51.618 -> Samples:42 min:89.24 max:90.13 time:748
12:22:51.618 -> x: -14868 y: 7020 z: 100
12:22:51.618 -> Tilt: 89.70
12:22:51.618 -> Tacc: 42.36
12:22:51.618 -> Volt: 4.26
12:22:51.618 -> Temp: 25.88
12:22:51.618 -> Gravity: 33.41
12:22:51.618 -> IP: 192.168.0.181
12:22:51.655 -> 
12:22:51.655 -> calling Ubidots
12:22:51.655 -> {"tilt":89.69512,"temperature":25.875,"battery":4.264859,"gravity":33.41217,"interval":20,"RSSI":-83}
12:22:51.836 -> Sender: Ubidots posting
12:22:51.836 -> POST /api/v1.6/devices/iSpindel000?token=xxx HTTP/1.1
12:22:51.836 -> Host: things.ubidots.com
12:22:51.836 -> User-Agent: ESP8266
12:22:51.836 -> Connection: close
12:22:51.836 -> Content-Type: application/json
12:22:51.836 -> Content-Length: 101
12:22:51.836 -> 
12:22:52.338 -> HTTP/1.1 200 OK
12:22:52.338 -> Server: nginx
12:22:52.338 -> Date: Thu, 10 Jun 2021 10:22:52 GMT
12:22:52.338 -> Content-Type: application/json
12:22:52.338 -> Transfer-Encoding: chunked
12:22:52.338 -> Connection: close
12:22:52.338 -> access-control-allow-origin: *
12:22:52.338 -> 
12:22:52.338 -> c0
12:22:52.338 -> {"battery":[{"status_code":201}],"gravity":[{"status_code":201}],"interval":[{"status_code":201}],"rssi":[{"status_code":201}],"temperature":[{"status_code":201}],"tilt":[{"status_code":201}]}
12:22:52.449 -> 0
12:22:52.449 -> 
12:22:52.449 -> 
12:22:52.449 -> Final-sleep: 20s; RT: 1903
[...]
```

peterschrott, 2021-06-10:
Oke, after half almost a day of debugging I am even more confused.

By adding some debug infos in the source code of 7.0.0 I found out, that the wifi status (`WiFI.status()`) is `7 – WL_DISCONNECTED` right after wakeup in `setup()`. Setting the wifi mode to `WIFI_STA` (iSpindel.cpp:1254) lets the wifi status change to `0 – WL_IDLE`. After the "rescue" the status is correctly `3 WL_CONNECTED`. The esp goes to sleep with this version, wakes up with status `7` ...

So I wanted to add the same debug infos to v 6.5.1 to, because the pre-compiled version worked just fine for me. I checked out the source by the tag `tag/6.5.1` and added the same logging as above. After compiling `pio run -e d1_mini` and flashing, I found out that the self-compiled fw suddenly has the same wifi issue as I encountered on the pre-compiled 7.0.0.

So I tried all previous version pre-compiled and self-compiled (macOS 11.2.3):
```
version   pre     self
7.0.0     ❌       ❌
6.6.0     ❌       ❌
6.5.1     ✅       ❌
6.5.0     ✅       ❌
6.2.0     ✅       ❌
```
Whenever I compile the bin myself this wifi connection issue pops up, regardless the version. The issue also appears from pre-compiled 6.6.0.

I have a vague hypothesis now: I see, that from version 6.6.0 the compilation is done with GitHub Actions. I assume before it was done on @universam1 machine. So, maybe there was some older version of a dependencies laying around on his machine which was then bundled. With the GH Actions its always a clean setup, as it was on my machine now. 🤷

Side note: version < 6.5.1 did not compile on my machine:
```
pio/src/iSpindel.cpp:649:28: error: cannot convert 'String' to 'bool' in return
  649 |       return sender.sendTCP("ispindle.de", 9501);
      |              ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
      |                            |
      |                            String
```
Here I am asking myself, how fw releases could have been built.

pppedrillo, 2021-06-10:
UPD all good with espesssif 8266 v 3.0.0 (the latest) as well

peterschrott, 2021-06-10:
@pppedrillo, yes I also did a clean build, deleting the mentioned directories. Also global pio caches.

I checkout the tag 6.5.1 (`git checkout tags/6.5.1 -b tag-6.5.1`) and build it with (`pio run -e d1_mini`). This leads to the compile error I posted above. In fact `sender.sendTCP("ispindle.de", 9501)` returns a `string`, see: https://github.com/universam1/iSpindel/blob/6.5.1/pio/src/iSpindel.cpp#L649
But `bool uploadData(uint8_t service)` (https://github.com/universam1/iSpindel/blob/6.5.1/pio/src/iSpindel.cpp#L542) expects a `boolean` to be returned. Thats what the compile error says.
I fixed it by processing the udp response like in: https://github.com/universam1/iSpindel/blob/6.5.1/pio/src/iSpindel.cpp#L654.

I am not sure what espesssif is. The version of the d1 mini? Is it relevant for compiling?

pppedrillo, 2021-06-10:
@peterschrott  
Indeed you right, It was broken in 6.5.1 and fixed in 6.6.0 (commit # 9872c32242b9f1f4021c8967e4c0d8798b18634d)

espesssif 8266 is a platform which is using for build ispindel image. 
https://docs.platformio.org/en/latest/platforms/espressif8266.html

peterschrott, 2021-06-10:
@pppedrillo oke, as I found out, it was broken for older versions too. Thats why I got confused and was wondering how those old releases could have been built.

Espressif is nothing I need to worry about when I build the fw on my machine? The way I described above is just fine, or does it need some more config or parameters?

Still, I can fix the compilation error for the old versions but still get this weird wifi behavior with my builds, the same as with pre-builds >= 6.6.0.

I made a clean build of the current master with some additional logging regarding wifi status in `setup()`: https://we.tl/t-wM5KmbAdgT 
Maybe you could flash your iSpindel and provide me the logs after?

I still don't think its about my d1 mini, as the old precompiled versions work find. Software... 😪

pppedrillo, 2021-06-11:
@peterschrott 
>Thats why I got confused and was wondering how those old releases could have been built.

Right, very confusing. May be it was guarded out by preprocessor with #if / #ifdef or such.... 

> Maybe you could flash your iSpindel and provide me the logs after?

<details>
  <summary>
Here you are. It is indeed sending data only once on start.
</summary>
<pre><code>
d▒▒▒$`▒g▒d
          FW 7.0.0
2.2.2-dev(38a443e)
Wifi status on wakeup: 7
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"ispindel000","Token":"xxx","Sleep":30,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"xxx","PSK":"xxx","Offset":[0,0,0,0,0,0]}
offsets not available
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Wifi status after setting STA mode: 0
Samples:42 min:56.28 max:69.95 time:752
x: -1812 y: 14640 z: 9710
Tilt: 56.82
Tacc: 30.42
Volt: 4.09
Temp: 27.56
Gravity: 16.60
Wifi status 'not connected': 0
After waiting 1ms, result wifi status: 0
Failed to connect -> trying to restore connection...
Wifi status: 0
Rescued Wifi credentials
   -> waited for 3924ms, result 3
   -> Connection restored!
                          Final-sleep: 30s; RT: 5001

d▒ǜl`▒▒g▒d
          FW 7.0.0
2.2.2-dev(38a443e)
Wifi status on wakeup: 7
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"ispindel000","Token":"xxxx","Sleep":30,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"xxx","PSK":"xxx","Offset":[0,0,0,0,0,0]}
offsets not available
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Wifi status after setting STA mode: 0
Samples:42 min:68.76 max:87.52 time:751
x: -14628 y: 8030 z: 746
Tilt: 87.39
Tacc: 29.74
Volt: 4.10
Temp: 27.50
Gravity: 32.25
Wifi status 'not connected': 0
After waiting 0ms, result wifi status: 0
Failed to connect -> trying to restore connection...
Wifi status: 0
Rescued Wifi credentials
   -> waited for 3923ms, result 3
   -> Connection restored!
                          Final-sleep: 30s; RT: 5003

$▒▒▒l ▒▒g▒d
           FW 7.0.0
2.2.2-dev(38a443e)
Wifi status on wakeup: 7
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"ispindel000","Token":"xxxx","Sleep":30,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"xxx","PSK":"xxx","Offset":[0,0,0,0,0,0]}
offsets not available
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Wifi status after setting STA mode: 0
Samples:42 min:87.37 max:87.49 time:751
x: -14618 y: 8012 z: 732
Tilt: 87.43
Tacc: 29.19
Volt: 4.10
Temp: 27.44
Gravity: 32.28
Wifi status 'not connected': 0
After waiting 0ms, result wifi status: 0
Failed to connect -> trying to restore connection...
Wifi status: 0
Rescued Wifi credentials
   -> waited for 3924ms, result 3
   -> Connection restored!
                          Final-sleep: 30s; RT: 5000

</code></pre>
<details>

zfeges, 2021-06-11:
I have the same problem, Ispindel sends the data only the first time, and later nothing.  That problem is from V6.6.0
6.5.1 work great!

peterschrott, 2021-06-12:
@pppedrillo thanks for the logs. This is whats happens on my iSpindel.

I have done a clean-clean build by using Vagrant with a clean Ubuntu 20.04 LTS, fresh checkout of the source, and fresh install of platformio (quite similar, what on Github Actions happens). 
=> Outcome: Same result as described above, Wifi issues on every wakeup, no data transfer.

@pppedrillo can you actually confirm, that you don't see this issue (data are uploaded) with the pre-build 7.0.0?

pppedrillo, 2021-06-12:
@peterschrott If I build firmware myself, 7.0.0 uploads data to Ubidots pretty well. No such issues like with your image.

<details>
 <summary>
Here is my log for couple of wakeups.
</summary>
<pre>

Final-sleep: 20s; RT: 1756
{$lܟ<▒l▒<
▒$▒c|ǃ▒{▒c▒
#▒▒'o▒dog▒▒▒cx▒▒l{l{$p▒g▒▒$
▒▒
co▒|▒d▒
▒c▒▒g'▒l▒▒l`▒gl n{▒▒▒gb$`;▒▒obl`▒#▒▒l{▒ ▒o▒
                                           FW 7.0.0
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel000","Token":"BBFF-qm9yKApVJaL1RZCZ2bjkdwqsAKE4KE","Sleep":20,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"xxx","PSK":"xxx","Offset":[0,0,0,0,0,0]}
offsets not available
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:89.05 max:89.16 time:755
x: 7996 y: 16368 z: 294
Tilt: 89.12
Tacc: 30.64
Volt: 3.98
Temp: 28.69
Gravity: 33.13
IP: 192.168.2.152

calling Ubidots
{"tilt":89.12337,"temperature":28.6875,"battery":3.983316,"gravity":33.12539,"interval":20,"RSSI":-79}
                      Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel000?token=xxxxx HTTP/1.1
Host: industrial.api.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 102

HTTP/1.1 200 OK
Server: nginx
Date: Sat, 12 Jun 2021 16:45:57 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
access-control-allow-origin: *

[......]

Final-sleep: 20s; RT: 3718
rdd▒▒|▒$▒|
▒
l▒
#|▒▒▒;▒c▒c▒▒og▒l'o▒▒▒cp▒▒dsdrd8▒o▒
▒
d▒▒
c
o▒|▒l▒
▒b▒▒og▒$▒▒l`▒n
l`o;▒▒▒g
cl`;▒▒gb$`▒#
▒▒ls▒`▒o▒
         FW 7.0.0
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel000","Token":"BBFF-qm9yKApVJaL1RZCZ2bjkdwqsAKE4KE","Sleep":20,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"xxx","PSK":"xxx","Offset":[0,0,0,0,0,0]}
offsets not available
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:89.05 max:89.16 time:757
x: 7986 y: 16370 z: 280
Tilt: 89.11
Tacc: 30.29
Volt: 3.97
Temp: 28.50
Gravity: 33.12
IP: 192.168.2.152

calling Ubidots
{"tilt":89.1097,"temperature":28.5,"battery":3.967675,"gravity":33.11853,"interval":20,"RSSI":-80}
                  Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel000?token=xxxxx HTTP/1.1
Host: industrial.api.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 98

HTTP/1.1 200 OK
Server: nginx
Date: Sat, 12 Jun 2021 16:46:19 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
access-control-allow-origin: *

[......]

Final-sleep: 20s; RT: 1714

</pre>
</details>

pppedrillo, 2021-06-12:
@peterschrott Have you tried completely wipe your wemos module? 
1. Flash 4 1Mb blanks. The last blank (started from 0x300000) will finish with error . its ok. Power module off and back on.
2. 
2.

peterschrott, 2021-06-14:
@pppedrillo thanks for the info & the logs. 

Locally I work with macOS. For the clean build I used Ubuntu to be analog with the Github Actions.

Now I build the firmware on Windows 10 (quite clean, fresh python and such). Unfortunately I encounter the same issues with my builds as before. In locally build v 6.5.1 and in v 7.0.0 I get that wifi issue on each wakeup.

So I still don't have a trace why the pre-builds > 6.5.1 do not properly work on my iSpindel. @zfeges reported the same issue.

Maybe @pppedrillo you could provide me you locally build v 7.0.0 (and if its not too much to ask also 6.5.1). Thanks!

pppedrillo, 2021-06-14:
@peterschrott  Here you are https://www88.zippyshare.com/v/KKxXBRJX/file.html

zfeges, 2021-06-14:
And I'm not the only one who has the same problem ... I know a dozen other people with the same problem.
After saving the settings, ispindel sends the data only for the first time.  Later no more ... I tested via wifi relay too ... same thing.

peterschrott, 2021-06-14:
> @peterschrott Here you are https://www88.zippyshare.com/v/KKxXBRJX/file.html

@pppedrillo, can't open: 403. Sorry.

pppedrillo, 2021-06-15:
https://www.dosyaupload.com/8e8j
@peterschrott

pppedrillo, 2021-06-15:
> And I'm not the only one who has the same problem ... I know a dozen other people with the same problem.
And it started only after update to 7.0.0?

zfeges, 2021-06-15:
Yes, after update and adding again configuration data... ispindel send data only first time. If I restart ispindel - nothing... when I go in management mode and exit... ispindel again send first time and after that 0...

pppedrillo, 2021-06-15:
@zfeges Have you tried to wipe wemos before flashing the image?

eg
1. flash blank 1MB image x4 times (from 0x0 , 0x100000, 0x200000 and 0x300000 )
2. flash internal://nodemcu from 0x0 + internal://blank from 0x1000
and after those flash your  image. 
The first wemos module's start might take a bit after that because it will prepare a new filesystem, but you'll have brand new fresh wemos module with firmware image after that.

peterschrott, 2021-06-15:
@zfeges didn't you say, that pre-build version 6.6.0 also did not work out for you? Maybe you can "confirm" my findings below?

@pppedrillo thanks a lot! You could always use https://wetransfer.com/ – which I used sending you my bin. No mail needed you can just create a link.

I have have tried your builds, and surprise: Both of them work like charm. To sum it up:
```
v         pre-build   peter     ppp.
7.0.0     ❌           ❌        ✅
6.5.1     ✅           ❌        ✅
```

And it must be about the build itself or the build environment. Especially since pre-build bins >= 6.6.0 are using Github Actions. This is the only significant change I can see in this repo. Unfortunately I am not deep enough in building firmware and the build tools. E.g. how about dependencies are handled. I also had the python or platformio version in mind. I am using Python 3.9.5 and pio  5.1.1. 

But it got even more interesting:
I followed your advice to wipe the flash. And suddenly the "wifi error" on the wakeup was different. After ONE successful connection & transmit of data to ubidots I see the following in the logs:
```
[...]
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 3.28
*WM: seconds
*WM: Connection result:
*WM: WL_CONNECTED
saving config...
{"Name":"iSpindel000","Token":"BBFF-2gObDxJcRstAGoqNfbNOorCvsNEbaj","Sleep":15,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"___erster_low","PSK":"17-Karotten","Offset":[0,0,0,0,0,0]}
saved successfully
Samples:32 min:85.12 max:86.39 time:546
x: 15292 y: 7060 z: 1264
Tilt: 85.79
Tacc: 41.49
Volt: 4.22
Temp: 24.81
Gravity: 31.45
IP: 192.168.0.181

calling Ubidots
{"tilt":85.78915,"temperature":24.8125,"battery":4.223149,"gravity":31.44902,"interval":1,"RSSI":-72}
Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel000?token=BBFF-2gObDxJcRstAGoqNfbNOorCvsNEbaj HTTP/1.1
Host: industrial.api.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 101

HTTP/1.1 200 OK
Server: nginx
Date: Tue, 15 Jun 2021 20:41:41 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
access-control-allow-origin: *

c0
{"battery":[{"status_code":201}],"gravity":[{"status_code":201}],"interval":[{"status_code":201}],"rssi":[{"status_code":201}],"temperature":[{"status_code":201}],"tilt":[{"status_code":201}]}
0

Final-sleep: 1s; RT: 28945
{l␀l��|␀�l�<␃␌␌␄�␌d�␌#|��␃�␓�;�c�␌c��og�$g'���␄c␜x��l;l{$p�g�␐␃␌␄�␌d␌��␌␌␄c␌n�|␃�␌Ǐ␄c��g'�␀l��l`␃�␓␛og␄�␏$␏␃'{���o␌␄;s�`␂8�o�␘␃␌␄{����␜��␃␄c␌n�|␃␄�␄c��og�␀␄�d`␂�␛␛'o␌�␇l␇␃orǛ�g␄␌s␛l ␃␏r��g␄␌s␛l`␃�␂␄␄␌d��␄l ␃��n�␏$
FW 7.0.0
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel000","Token":"BBFF-2gObDxJcRstAGoqNfbNOorCvsNEbaj","Sleep":15,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"___erster_low","PSK":"17-Karotten","Offset":[0,0,0,0,0,0]}
offsets not available
Boot-Mode: Deep-Sleep Wake
...........
ERROR no Wifi credentials

going to Config Mode
reboot RFCAL
{l␀l��|␀�l�<␃␌␌␄�␌d�␌b|��␃�␓�r�#�␄c��og�lgn���␄c␜8��l;l{lp�g�␐␃␌␄�␌d␌��␌␌␄c␌n�|␃�␌Ǐ␌c��gn�␀l��l`␃�␛␓'o␌�␇d␇␃orǛ�g␄␌s{�`␃p�g�␐␃␌␄;����␜��␃␄c␌n�|␃␄�␌b��no�␀␄�l`␃�␓␛og␄�␏$␏␃gs�ۓn␌␄{␛l`␃␇s��'␄␌s␛l`␃�␃␄␌␄$��␌l`␃��n�␏$
FW 7.0.0
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel000","Token":"BBFF-2gObDxJcRstAGoqNfbNOorCvsNEbaj","Sleep":15,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"___erster_low","PSK":"17-Karotten","Offset":[0,0,0,0,0,0]}
offsets not available
Boot-Mode: Deep-Sleep Wake

Double Reset detected
...........
ERROR no Wifi credentials

going to Config Mode
[...]
```

The "error" is different now – it says `ERROR no Wifi credentials` which happens according to the code, when the `WiFI` module (or how you call it in cpp) does not have an SSID. This happens on pre-builds v 7.0.0 and also my local builds (v 6.5.1).

Initially i flashed v 6.5.1 and then upgraded to 7.0.0 without wiping. Maybe I did not run into that `ERROR no Wifi credentials` because the _"wifi credentials were somehow stored from the old version"_. After wiping the flash this is not the case anymore. So in my upgrade case the d1 was just not connected anymore – but still had the SSID somewhere– after the sleep. (Sorry, some assumptions, as I am not too deep into embedded systems – just sharing my thoughts).

My conclusion so far: The `WiFI` somehow forgets the wifi credentials on sleep with some builds.

peterschrott, 2021-06-15:
@pppedrillo found it! 🎉

The persistence of wifi has changed in `esp8266wifi`: https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html#persistent

It recommends to always use `Wifi.begin(..)`, but to restore the old behavior you can just call: `enableWiFiAtBootTime()`.

After adding this in my `setup()` my local builds work like charm. I ll open up an pr later this week.

Side question: Where do I define the version of that dependency? Is it somehow tied up with platformio?

pppedrillo, 2021-06-16:
> Unfortunately I am not deep enough in building firmware and the build tools. E.g. how about dependencies are handled.

@peterschrott You can just delete .pio folder in your project folder, it will re-download all the dep libs.

> I followed your advice to wipe the flash. And suddenly the "wifi error" on the wakeup was different. 

That error happens if you skip N1 "flash blank 1MB image x4 times (from 0x0 , 0x100000, 0x200000 and 0x300000 ) (I assume you have 4MB module)" and start directly from N2 (internal:// nodemcu, blank)

Please note, what if you have 4MB module, the last file flashing (from 0x300000) will end up with error. This is ok.

[image]

In case you don't have/can not generate blank file yourself, I attached a  good one.
[blank_1MB.zip](https://github.com/universam1/iSpindel/files/6661429/blank_1MB.zip)

> It recommends to always use `Wifi.begin(..)`, but to restore the old behavior you can just call: `enableWiFiAtBootTime()`.

Funny enough, I have just wifi.begin and it works for me without problem. I have seen "ERROR no Wifi credentials", but it was gone after full wiping as described above.

> Side question: Where do I define the version of that dependency? Is it somehow tied up with platformio?

Yes. Should be in **platformio.ini** in root folder of your project.

**lib_deps =
  ArduinoJson@>5
  ESP Async WebServer
  I2Cdevlib-MPU6050
  OneWire
  DallasTemperature
  RunningMedian
  PubSubClient
  Blynk
  ThingSpeak**

peterschrott, 2021-06-16:
This time I was doing the wipe before flashing. So the "real" issue came out. I am using `esptool` (Python lib, cmd line) which provides easy way to erase flash.

> Good catch, Funny enough, I have just wifi.begin and it works for me without problem. I have seen "ERROR no Wifi credentials", but it was gone after full wiping as described above.

I was more talking about, that on every `setup()` one should work wit `WiFi.begin()` to reenable wifi because the persistence of the wifi module has changed. The current implementation of the iSpindel relies on the automatic reenabling of the wifi. Thats way I added (PR is open) `enableWiFiAtBootTime()` to fix it. But in a long rund the handling of the wifi after sleep should be reworked. 

> And I have Arduino 5.8 I believe. 
Where is the Arduino version defined?

zfeges, 2021-06-16:
> 
> 
> @zfeges Have you tried to wipe wemos before flashing the image?
> 
> eg
> 
>     1. flash blank 1MB image x4 times (from 0x0 , 0x100000, 0x200000 and 0x300000 )  (I assume you have 4MB module)
> 
>     2. flash internal://nodemcu from 0x0 + internal://blank from 0x1000
>        and after those flash your  image.
>        The first wemos module's start might take a bit after that because it will prepare a new filesystem, but you'll have brand new fresh wemos module with firmware image after that.

I did not try on a blank wemos, it is my friend and he had a problem that his wemos did not save his data after entering.
I will try it in a few days on the new wemos.

peterschrott, 2021-06-18:
PR (https://github.com/universam1/iSpindel/pull/513) is merged into master. And will be available in the next fw release.

-> Therefore closing this issue.

closed 2021-06-18 by peterschrott

pppedrillo, 2021-06-30:
@peterschrott For what it worth,  I just noticed the logs you posted [here]( https://github.com/universam1/iSpindel/issues/507#issuecomment-861825135 )  have token for your API as well as ssid and psk password to your wifi network. Just in case you may want to remove it.

peterschrott, 2021-06-30:
@pppedrillo thanks, was a late night post. I edited the post, its still visible in the history, i ll revoke my ubidots token.

joeleonjr, 2023-08-30:
Hi! I'm a security researcher @TruffleSecurity. I found a valid API key leaked in a comment in this Issue. I highly recommend rotating the API key immediately (not just deleting the comment). Here's a blog post I wrote on remediating these types of vulnerabilities: https://trufflesecurity.com/blog/remediate-leaked-api-keys-with-key-rotation.<br><br>If you can't find the leaked API key, you can use the following [TruffleHog](https://github.com/trufflesecurity/trufflehog) command to automatically scan for it:<br>`trufflehog github --repo https://github.com/<your_username>/<your_reponame> --issue-comments --only-verified`<br><br>If it's not clear where/what the security issue is, please let me know. I'd be happy to help. Please note that the API key might be in the past edit of a comment.<br><br><br><br>(cc @pppedrillo)