universam1/iSpindel #512: ISpindel doesn't work with gyros which test OK with wire.h

Colin66, 2021-06-11:
Bug report.

iSpindel (tested with F/w 6.2 and 7.0) doesn't see any of a large batch of gyros & reports "Acc Test Connection ERROR!"

When I test the gyros with an Arduino sketch to scan addresses using wire.h they show up correctly on 0x68 and when I query them with wire.write (code as below) they appear to return valid values, e.g. :

AcX = 65 | AcY = -28583 | AcZ = 13966 | Tmp = 78.57 | GyX = 22354 | GyY = 6463 | GyZ = 29868

ISpindel code uses MPU6050.h and MPUoffset.h  
Is there a compatibility issue between these MPU libraries and some (many..most?) generic MPU6050 GY521s ?
If so can they be dropped from the code and use wire.h instead for improved compatibility ?

The Issues log has similar reports and online discussions widely report non-working gyros, possibly due to the same compatibility issue.
Thanks.

~~~~~
Code used to pull the values :  

  Wire.beginTransmission(MPU_addr);
  Wire.write(0x6B);  // PWR_MGMT_1 register
  Wire.write(0);     // set to zero (wakes up the MPU-6050)
  Wire.write(0x3B);  // starting with register 0x3B (ACCEL_XOUT_H)
  Wire.endTransmission(false);
  Wire.requestFrom(MPU_addr,14,true);  // request a total of 14 registers
  AcX=Wire.read()<<8|Wire.read();  // 0x3B (ACCEL_XOUT_H) & 0x3C (ACCEL_XOUT_L)    
  AcY=Wire.read()<<8|Wire.read();  // 0x3D (ACCEL_YOUT_H) & 0x3E (ACCEL_YOUT_L)
  AcZ=Wire.read()<<8|Wire.read();  // 0x3F (ACCEL_ZOUT_H) & 0x40 (ACCEL_ZOUT_L)
  Tmp=Wire.read()<<8|Wire.read();  // 0x41 (TEMP_OUT_H) & 0x42 (TEMP_OUT_L)
  GyX=Wire.read()<<8|Wire.read();  // 0x43 (GYRO_XOUT_H) & 0x44 (GYRO_XOUT_L)
  GyY=Wire.read()<<8|Wire.read();  // 0x45 (GYRO_YOUT_H) & 0x46 (GYRO_YOUT_L)
  GyZ=Wire.read()<<8|Wire.read();  // 0x47 (GYRO_ZOUT_H) & 0x48 (GYRO_ZOUT_L)
  Serial.print("AcX = "); Serial.print(AcX);
  Serial.print(" | AcY = "); Serial.print(AcY);
  Serial.print(" | AcZ = "); Serial.print(AcZ);
  Serial.print(" | Tmp = "); Serial.print(Tmp/340.00+36.53);  //equation for temperature in degrees C from datasheet
  Serial.print(" | GyX = "); Serial.print(GyX);
  Serial.print(" | GyY = "); Serial.print(GyY);
  Serial.print(" | GyZ = "); Serial.println(GyZ);
  Wire.endTransmission(true)

~~~~~
code used to scan for device addresses 
{
  Wire.begin(0,2);
  Serial.begin(9600);
}
void loop()
{
  byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
      nDevices++;
    }
etc...
~~~~~~

pppedrillo, 2021-06-12:
iSpindel checks WHO_AM_I register and if it is != 0x34 - it output this error. 
0x68 looks good
[image]

Sounds like a dodgy PCB. What PCB are you using?

Colin66, 2021-06-16:
Hi, not using a pcb, using gyros soldered directly to wemos.  Rest of circuit working fine, eg ds18b20 recognised correctly.
I'll try reading that register and post the result.
I don't know what MPU6050.h and MPUoffset.h do differently than wire.h as my programming is limited, but i can communicate with the gyros with wire.h and ispindel can't.  
Many people seem to have similar problems with compatibility of generic / cheap gyros.

peterschrott, 2021-06-18:
Just to close the loop:
It seems there are a lot of gy521 out there who do not follow the specs and do not send `0x68` on `WHO_AM_I`.

There are a few issues:
- https://github.com/universam1/iSpindel/issues/466
- https://github.com/universam1/iSpindel/issues/459
- https://github.com/universam1/iSpindel/issues/429
- ...

Most people say, that up to 50% of the gyros are just broken.. 

The mpu6050 lib was already extend by an second who am i response:
https://github.com/ElectronicCats/mpu6050/pull/25/commits/5073f03f53e6429cff98459aeb180e178db9098c

I have the same issue with my gyro, see the `Acc Test Connection ERROR!` but x, y, z values can be calculated. (my who am i code 0xC) @Colin66 you are right, the implementations are not aligned across the codebase. I am not sure how much "broken" the gyro itself is if it sends the wrong who am i code and if it could be still used by "ignoring" the who am i code and anyways fetch data.

Colin66, 2021-06-18:
Is there a need for the identification of the attached device in the iSpindel ? By design it's present and data will come from 0x68 from all gyros including the common non-compliant ones
Could it all be done with wire.h as in my sketch above ?

pppedrillo, 2021-06-18:
@Colin66 
Pls read this
https://github.com/universam1/iSpindel/issues/429#issuecomment-761690760

pppedrillo, 2021-06-18:
> I am not sure how much "broken" the gyro itself is if it sends the wrong who am i code and if it could be still used by "ignoring" the who am i code and anyways fetch data.

@peterschrott Are you sure that  these x,y and  z values are correct? And are they always correct for every single reading? 
Those "bad"/"broken"  gyros usually have a problems with calibration as well. Simply put, you either could not calibrate them or polynomial will be crazy. Also, iSpindel averaging gyro data from several readings to end up with result. 
In any cases with a gyro that has a broken who_am_i register you most likely never get correct gravity values.
iSpindel requires quite precise and consistent data/values from gyro, thats why it has been decided to keep away from such dodgy accel modules.

peterschrott, 2021-06-18:
@pppedrillo, no I don't know if they are correct. I just see some delivery, but as I said: I don't know how broken they are. But I guess, no one could say. The wrong WHO AM I is an indicator that it is somehow broken.

I just wanted to point @Colin66 to those discussions if she/he is interested in reading up. 

For myself: I just both 5 at once and i ll find out which one works.

pppedrillo, 2021-06-18:
> The wrong WHO AM I is an indicator that it is somehow broken.

Exactly! 

> For myself: I just both 5 at once and i ll find out which one works.

Right. Myself I usually buy from trusted sources or, if it is just random seller on Aliexpress, I always test them with Arduino/bread board before soldering. https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050

Usually if 1 of the batch is dodgy, then all the rest are the same crap as well. :(

Colin66, 2021-06-19:
@peterschrott  @pppedrillo  yes there are lots of posts with people having the same problem.
Wrong whoami suggests many of these parts are not compliant to the standard, but they appear to work Ok and a code tweak could allow the ispindel to communicate with them.  Seems like the checks could be removed to allow more people to build successfully and get a working unit.

pppedrillo, 2021-06-19:
> but they appear to work Ok

@Colin66 
Have you tried to disable  this check and test with yours?  
Are you able to calibrate it? 
After calibration, is iSpindel returning the correct gravity readings for the whole calibrated interval?

peterschrott, 2021-06-19:
@Colin66, I was also quite frustrated about the broken gyro after assembling my first one. Esp. as it is soldered there and can not be easily removed and not easy tested if you don't have the tools. Thought about adjusting the fw (similar as you proposed above) and using my build of it.

But after reading more about it I ended up just buying 5 of them. Especially, as @pppedrillo said: one can never know if it provides continously correct values. 

Those checksums are there to ensure the proper functioning of those kind of parts. And to go with @pppedrillo again: It will never be "user-friendly" if all WHO AM Is are accepted.

Not only to ensure a user-friendly use of the iSpindle for me the user-friendly assembling is the problem here. Two points:
1) I did not know about broken gyros, so I obviously bought only 1. If there was a hint to buy a bunch, i would have done so. 
2) Even I had bought some stock gyros, I would not have been able to test them before soldering there.

Fist issue can be solved by just ordering again - might be annoying to wait and pay again the shipping fee. But I am still kinda clueless how to "user-friendly" test those lille brats. 🤷

To "sove" those issue, the readme here could be updated with an hint to just buy more of the gyros and if someone who has an idea for a straight forward way to test the gyro before soldering this could be added to. I guess this would be an appropriated solution to avoid frustration.

ErikdBr, 2021-06-19:
> someone who has an idea for a straight forward way to test the gyro before soldering

The most straight forward or easy way to test is what I described here: https://github.com/universam1/iSpindel/issues/429#issuecomment-766389383
But you going to need some stuff
Also the post directly after mine of @thegreatgunbantoad could help.

peterschrott, 2021-06-20:
@ErikdBr thanks for the details test description. As you say, additional stuff is needed which is not at stock for most the iSpindel builders. 
I don't really understand what @thegreatgunbantoad was trying to say.

ErikdBr, 2021-06-20:
> I don't really understand what @thegreatgunbantoad was trying to say.

He has built one iSpindel without the wemos and gyro, just with the headers for them, so he can clip on and off without soldering the wemos and gyro in order to test these.

When you build iSpindels in bulk for fellow homebrewers these tests of gyros and also wemos(only nescesary if you use wemos clones,  I only use the originals, see wemos.cc) are paying for itself

peterschrott, 2021-06-21:
@ErikdBr cool, thanks for clarifying. Not an hw-guy here. How do you  "clip on and off without soldering"?
wemos.cc does not work for me, but looks like a temp issue.

ErikdBr, 2021-06-21:
https://www.wemos.cc/en/latest/d1/d1_mini.html works better. Directly beneath the pictures on that page there is a link 'Buy it' which will redirect to aliexpress. These are the original wemos(now Lolin) and the only ones where I did not have any issues with. Clones that I have tried because they where half the price, all had issues themselves and/or in combination with the gyro.

Just lay it loose on the headers and make sure they make good contact by pressing with your finger. This an example with a breadboard but it can be done with a partially build iSpindel like @thegreatgunbantoad did.
[image]

thegreatgunbantoad, 2021-06-21:
@peterschrott Does this make sense?  Solder some pins into the gyro or wemos and fit headers to a testing pcb.  There is no need to test temp sensors, they are almost never duff.

Ignore the small purple pressure sensor board.  It just doesn't really read sensibly when it's been tubed, was just an experiment that stalled as I had no way to properly test it over and above how hard I can blow.
[image]

peterschrott, 2021-06-21:
@ErikdBr thanks again for the explanation. I bought my from a trustworthy reseller in Germany (reichelt.de). Did not have a problem either – just crashed 2 out of stupidity. My corners are not round, they are flat which are according to [opensourcedistilling](https://www.opensourcedistilling.com/ispindel-troubleshooting/) bad. Buy my pins are also flipped from right left. So I had to solder it up side down.

@thegreatgunbantoad thanks 2! I could have just translated "headers" before asking dump questions. Sorry for that! I actually have in mind soldering the headers onto the board and just plug the gyro in for permanent use. I still could fit into the tube. For 2 reasons: I don't have all the spare parts 2) if I solder it there once I can no de-solder it. Do you think this is a good idea?

So basically this two options are very clear and straight forward. For 1-time-iSpindel-builders like me who just want to "buy the kit" and solder it together not really doable (no offence).

pppedrillo, 2021-06-22:
> My corners are not round, they are flat which are according to [opensourcedistilling](https://www.opensourcedistilling.com/ispindel-troubleshooting/) bad.

I have soldered bunch of wemos modules with "not round" corners bought from random sellers on Aliexpress. Never had a problems with them. Gyros - yes, sometimes. WemOs - no, always were fine. 
Things could got changed since then tho, last time I bought a batch in Feb 2021.

[image]

peterschrott, 2021-06-25:
Inspired by @thegreatgunbantoad I also soldered headers to my PCB and pins to the gyro. So I can just plug the gyro in and see if its working on a fully assembled iSpindel. 

1) First I clipped of 4 headers and filed them off – as otherwise the plugged in gyro could not fit into the tube.

[image]

2) I soldered the headers to my PCB & pins to the gyro accordingly.

[image]

3) I plugged in the gyro and could test it on my fully assembled iSpindel.

[image]

4) As the pins did not sit super tight in the headers the gyro was a little bit wiggly. So after I made sure the gyro is working and it can be calibrated I just used some superglue to fix it after adjusting it propperly. 

In case, the gyro was bad, I could have just plugged in another one. As the pins of it are under the battery holder de-soldering it is quite a pain and messes up the PCB if it needs to be done several times.

pppedrillo, 2021-06-26:
@peterschrott 
You can also use so-called "short" AKA "low" header - then everything fits the tube without filing and clipping.

[image]

peterschrott, 2021-06-26:
@pppedrillo very cool! Thanks. The high ones came with the d1 mini. :)

Xosted, 2021-07-05:
Hello everyone,
I tried reading the entire feed and I thought I would add my 2 cents to it.
**In short,** I think the check/ban is excessive and the situation could be easily enhanced with a warning for a better user experience. (suggestion details below)

The following isn't an attack on anyone. That is a user feedback from a newbie:
**In long,** I am facing what I believe is the exact same problem. Luckily for me, I built one unit that worked straight out, or almost. I didn't notice the "1 time update to ubidots" problem but now it is fixed with the latest release.
However, the second one hit me with the O tilt in the configuration. 
My experience of this whole endeavor is that... If that had been my very first device, I might have given up all together. I am not a pro of the field like many of you seem to be. I don't have your logic in what makes a good part or a bad one. I don't know what makes a reputable source or not. I ordered what I found on Amazon...
However, there probably must be a lot MORE copies out there than there are genuine parts. Yes, they might be unreliable, but they might work too. Not everyone has access to the same sourcing opportunities. A blanket ban while so many people seem to face this problem is not very user friendly. My suggestions is:

Remove the ban and replace it by a warning: The device already checks for the gyro's ID so it could certainly report not getting one (faulty part or solder), a wrong ID (then check if it gets values and tell the user that it might not be reliable) or the right ID and go on. That would allow people facing this to know if they have to go in with scissors or if they have to triple check the measurements at first to see if it's a bad but working part. If measurements are not good then no need to come here to ask as the red warning banner on the config page already told you.

(By the way, it is also valid for the "no sensor, no wifi" problem. Why not allow the wifi and config mode to get up and report to the user "the sensors are not found, check your soldering". Instead of a simple stuck device which could be due to anything?)

I believe this would provide a much better user experience, no?

closed 2021-09-10 by stale[bot]