universam1/iSpindel #517: ERROR no Wifi Credentials...parse error

delight-f, 2021-06-30:
## I'm submitting a ...
  - [x ] bug report
  - [ ] feature request
  - [x ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [ x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x ] Checked the logs for errors in the terminal console ?
- [x ] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

Please replace this line with a brief summary of your issue

### Console Logs

      FW 7.0.0
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel001","Token":"","Sleep":180,"Server":"192.168.2.142","API":3,"Port":80,"Channel":0,"URI":"/gravity","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Inst                                  ance":"000","Hassio":false,"Vfact":191.8,"TS":0,"OWpin":12,"POLY":"(1.0000E-6*(tilt^3))+(-1.3137E-4*(tilt^2))+(6.9680E-3*tilt)+(8.8885E-1)","SSID":"dd-wrt","PSK":"***","Offs                                  et":[-1480,-2024,378,-357,12,-35]}
applying offsets: -1480:-2024:378
confirming offsets: -1480:-2024:378
Boot-Mode: Deep-Sleep Wake

Double Reset detected
...........
           ERROR no Wifi credentials

going to Config Mode
*WM: Scan done
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: warning1
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM: hassio
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel_iSpindel001
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: After waiting...
*WM: 0.00
*WM: seconds
*WM: Timed out connection result:
*WM: WL_IDLE_STATUS
Rescued Wifi credentials
   -> waited for Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
5325ms, result 3
Samples:33 min:94.87 max:95.01 time:568
x: -12450 y: 5402 z: -1168
Tilt: 94.94
Tacc: 24.99
Volt: 3.99
Temp: 24.25
Parse error at 1
Gravity: 0.00
IP: 192.168.2.126

calling HTTP
{"name":"iSpindel001","ID":14435155,"angle":94.94474,"temperature":24.25,"temp_units":"C","battery":3.98853,"gravity":0,"interval":180,"RSSI":-63}HTTPAPI: posting
code: 200
{}

Final-sleep: 180s; RT: 309989

## Steps to Reproduce (for bugs)
1. Fresh install of 7.0.0
2. Add wifi credentials
3. Save

## Context
Using iSpindel as a sensor within Brewpiless 

## Your Environment

- Firmware Version used:7.0.0
- HTTP

pppedrillo, 2021-06-30:
you need completely [wipe ispindel]( https://github.com/universam1/iSpindel/issues/507#issuecomment-862164046 ) before install 7.0

ErikdBr, 2021-06-30:
> you need completely [wipe ispindel](https://github.com/universam1/iSpindel/issues/507#issuecomment-862164046) before install 7.0

Is that the case for 7.1.2 as well?

universam1, 2021-06-30:
@delight-f `Parse error at` means your polynomimal is broken, start fresh with default one to verify.

WiFi is apparently working, you got an IP `IP: 192.168.2.126` 

Upgrade to latest firmware, reset your device and report again if it persists

pppedrillo, 2021-06-30:
> 
> Is that the case for 7.1.2 as well?

If you have it flashed with FW before 7.0.0 (ie having SPIFFS) , without complete wipe you might experience a problems with FW >= 7.0.0 - wifi creds are stored in special sectors of flash memory which may need to be reinitialized.

delight-f, 2021-07-01:
> 
> 
> @delight-f `Parse error at` means your polynomimal is broken, start fresh with default one to verify.
> 
> WiFi is apparently working, you got an IP `IP: 192.168.2.126`
> 
> Upgrade to latest firmware, reset your device and report again if it persists

The polynomial was copied straight from the ispindelpolyestimator.xlsx
(1.0000E-6*(tilt^3))+(-1.3137E-4*(tilt^2))+(6.9680E-3*tilt)+(8.9306E-1)

Can you confirm how this should be formatted correctly?

closed 2021-09-04 by delight-f