universam1/iSpindel #518: Refactoring

schieska, 2021-07-01:
## I'm submitting a feature request

So, I am working on a fully SMD assembled iSpindel and I would like to add different sensors for increased accuracy and decreased power usage. But with the way this project is structured it is not that easy to add support for different sensors without breaking other stuff.

## Features
I suggest refactoring the code to accommodate faster development, easier feature development & options for different sensors.
Ideas that come to mind are, 

- generic functions for initializing/requesting data from sensors.
- splitting the code in multiple files

Maybe for some inspiration check out the way how marlin (3d printer software) is build.

## Why is this useful?
As I said earlier, code split in modules means rewriting without breaking other parts, easier to add new features, easier to support more sensors and readability.

I would love to help with this if you find this a good idea

pppedrillo, 2021-07-03:
Why don't you contribute then? :) Please go ahead :)

schieska, 2021-07-03:
Just want to make sure, I'm not the only one who would think this would be useful

Draco1557, 2021-07-04:
Such would be helpful but it doesn't just apply to this project which is why I didn't comment earlier - need to change the culture across the entirety of Github. Many, if not all, projects on Github are missing the standard documents (such as a UML diagram (https://www.smartdraw.com/uml-diagram/)) that would give people new to the code an understanding of how the code is structured.

thegreatgunbantoad, 2021-07-04:
There's a few functions that use variables that are not passed to the function too that could do with a tidy up.  Putting all the variables that get written to the json into a structure may be helpful as then intellisense in VS code will tell you what variables are in there. I did do a bit of work on that a while back I'll try and dig it out.

closed 2021-09-10 by stale[bot]