universam1/iSpindel #519: https or redirections

joellord, 2021-07-07:
Hi everyone,
   not sure if I just can't find the feature or if it's not existing. I want to make requests to a serverless endpoint that I have opened for POST requests. Right now, it seems like the requests from my iSpindel are sent via HTTP. I get a code 301 as a response which I believe is because it tries to redirect to the https page. 
   I am not very proficient with C, but looking at the code seems to indicate that it would be possible to send requests via https, I just can't seem to figure out how to configure my device to do so.

  Is there any way to send an HTTPS request? If not, this is a feature request :)

## I'm submitting a ...
  - [ ] bug report
  - [x] feature request
  - [ ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x] Checked the logs for errors in the terminal console ?
- [x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

pppedrillo, 2021-07-07:
bool SenderClass::sendHTTPSPost(String server, String uri)

joellord, 2021-07-07:
I saw that method, I just can't figure out how to configure the iSpindel to it ¯\_(ツ)_/¯

joellord, 2021-07-14:
I just realized that I did not upgrade to the lastest firmware. I can now see the HTTPSPost option in the configuration menu.

¯\_(ツ)_/¯

closed 2021-07-14 by joellord