universam1/iSpindel #527: Use of integrated solutions like smartbands

scientistnobee, 2021-07-20:
Hi, I found this project very interesting. I would like to know if it is possible to use either a M5Stick5 or ESP-watch than building everything from scratch. 

https://github.com/m5stack/M5StickC

https://www.aliexpress.com/item/4001162589218.html

FernandoGarcia, 2021-07-28:
Hi!

Looks like your question doesn't belong here.

Maybe you are talking about [Nautilis](https://www.nautilis.eu/en/portfolio-item/nautilis-ispindel-relay/) alternative. If so this not the right place to ask.

Best regards.

thegreatgunbantoad, 2021-07-28:
I don't think the OP is talking about the nautilus.

The wristband doesn't seem to have a temp sensor, but it has wifi and an accelerometer.  So it's possible you could use it for just the SG(tilit), but I don't know how much code you could borrow.

The https://shop.m5stack.com/products/m5stickc-plus-esp32-pico-mini-iot-development-kit looks very doable if you can add a temp sensor and waterproof it.  But again not sure how much code you can move across.

Definitely an interesting idea, I'd be more inclined to go down the M5Stick route as I think there is more support and info with that device.

scientistnobee, 2021-07-28:
That's true, M5stic doesn't have a temperature sensor, but it is easy to add through the grove port. Software seems to be the tough part.

FernandoGarcia, 2021-07-30:
I think that this type of question should be [here](https://github.com/universam1/iSpindel/discussions).

closed 2021-10-06 by stale[bot]