universam1/iSpindel #532: Button / page to force connection to wifi?

m0n5t3r, 2021-08-13:
## I'm submitting a ...
  - [x] bug report
  - [x] feature request
  - [ ] support request => Please use the [Discussion](https://github.com/universam1/iSpindel/discussions) board instead!

## Did you follow the general troubleshooting steps first:

- [x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x] Checked the logs for errors in the terminal console ?
- [x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

The problem: when starting in config mode, it seems the iSpindel (7.1.3, current master) will not connect to wifi most of the time, even if credentials are set correctly and it can send data in normal mode (influxdb in my case); after poking at it for a while, it seems to be pretty random:

* I thought it connects when the stored configuration changes (after many resets it seems it loses the sleep value and sets it to 1, and when setting it back to 1800 it connects to wifi when starting in config mode...)
* ... but then after managing to talk to it over wifi from the computer and changing the polynomial expression it failed to connect to wifi again
* ... then it worked again after changing the sleep value from 1800 to 1801, then failed when changing it from 1801 to 1800, then worked again when changing it from 1800 to 1802, so decreasing the sleep time seems to make STA mode fail, which makes no sense ans is probably not what is happening at all

Rather than embarking on a heisenbug chase, would it be better to add a button that says "connect to wifi"?

### Console Logs

logs are normal, no errors in there (and the status is just `*WM: SET AP STA` when wifi connects and `*WM: SET AP` when it doesn't);

failed session:

```
                                                                                                                                    FW 7.1.3
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel_Yellow","Token":"","Sleep":1800,"Server":"x.x.x.x","API":8,"Port":8086,"Channel":0,"URI":"","DB":"ispindel","Username":"ispindel","Password":"mypassword","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":308.77,"TS":0,"OWpin":12,"POLY":"0.000001548*tilt^3-0.000186440*tilt^2+0.008099901*tilt+0.905368620","SSID":"homewifi","PSK":"password","Offset":[-3654,1404,4868,26,18,-63]}
applying offsets: -3654:1404:4868
confirming offsets: -3654:1404:4868
Boot-Mode: Deep-Sleep Wake

Double Reset detected

going to Config Mode
*WM: Scan done
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: warning1
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM: hassio
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel_iSpindel_Yellow
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
```
successful session:

```
                                                                                                                                         FW 7.1.3
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel_Yellow","Token":"","Sleep":1802,"Server":"x.x.x.x","API":8,"Port":8086,"Channel":0,"URI":"","DB":"ispindel","Username":"ispindel","Password":"mypassword","Job":"ispindel","Instance":"000","Hassio":false,"Vfact":308.77,"TS":0,"OWpin":12,"POLY":"0.000001548*tilt^3-0.000186440*tilt^2+0.008099901*tilt+0.905368620","SSID":"homewifi","PSK":"password","Offset":[-3654,1404,4868,26,18,-63]}
applying offsets: -3654:1404:4868
confirming offsets: -3654:1404:4868
Boot-Mode: External System
power-cycle or reset detected, config mode

going to Config Mode
*WM: Scan done
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: warning1
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM: hassio
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP STA
*WM:
*WM: Configuring access point...
*WM: iSpindel_iSpindel_Yellow
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
```

## Context
this makes config changes tedious (need to use the phone or stop what I'm doing at the computer and connect to the iSpindel AP)

## Your Environment
- Firmware Version used: 7.1.3
- Service backend you have configured the iSpindel to talk to: influxdb

## Features
I'll try to figure out how to add the button / page when I have some time, if this is deemed a good idea.

ErikdBr, 2021-08-13:
Maybe I am completely missing the point here but when the iSpindel is in config mode it acts like a wifi access point and therefore the iSpindel itself does not connect to wifi. The whole point of this is that you can then connect to this AP with another device, phone/PC/whatever and go to the config portal at 192.168.4.1 and change whatever. 
After saving the config it resets itself and then connects to the wifi configured in the config page and will do so every "Sleep" seconds.

What I do find strange is that in your failed session it reads: "Double Reset detected going to Config Mode" and it sets up a HTTP server at 192.168.4.1 so this works as designed.
However in your successful session it reads: "power-cycle or reset detected, config mode going to Config Mode". It should not go to config mode after a power cycle or a single reset unless you didn't fill in the right wifi credentials ("SSID":"homewifi","PSK":"password" in your log) if so it will go to config mode because it doesn't know to which SSID it should connect.

Hope this helps?

m0n5t3r, 2021-08-13:
it's supposed to connect to wifi if it has it configured as far as I understand; it was a reset from config mode, which apparently makes it restart in config mode

the values in this ticket are obfuscated, but they are correct in the config (in normal mode it sends data to influxdb just fine, apart from the first run after powering up when it spends about 4s "rescuing" (or maybe recovering, don't remember the word and I've put the things aside for now) something about wifi and then goes to sleep)

to be honest, I fail to understand what [this](https://github.com/universam1/iSpindel/blob/master/pio/src/iSpindel.cpp#L408) is calling, because there is no such signature [here](https://github.com/universam1/iSpindel/blob/master/pio/lib/WiFiManagerKT/WiFiManagerKT.cpp#L325) and there seems to be some other magic in there (who tells the wifi to connect?)

m0n5t3r, 2021-08-13:
https://github.com/universam1/iSpindel/blob/01bcd8ae02c636eddf739ae97a3f3c2225b6c864/pio/src/iSpindel.cpp#L243 - no idea what those words mean right now, but that variable seems to trigger config mode

ErikdBr, 2021-08-13:
> https://github.com/universam1/iSpindel/blob/01bcd8ae02c636eddf739ae97a3f3c2225b6c864/pio/src/iSpindel.cpp#L243
> 
> - no idea what those words mean right now, but that variable seems to trigger config mode

In the comment above that boolean it states: "// I added a single reset detection as workaround to enter the config-mode easier"
I don't know better than that it should not enter config mode on a single(press) reset or after a save of the config, it should restart and nothing else.
It should only go into config mode on a double reset(press the reset button fast twice or more) . I think this is the problem.

Anyone else has a comment on this?

pppedrillo, 2021-08-13:
> to be honest, I fail to understand what [this](https://github.com/universam1/iSpindel/blob/master/pio/src/iSpindel.cpp#L408) is calling, because there is no such signature [...]

No magic. Signature is in [place](https://github.com/universam1/iSpindel/blob/3312f2fa5e944a80481c36a37ece19fac91c24b9/pio/lib/WiFiManagerKT/WiFiManagerKT.cpp#L331)

m0n5t3r, 2021-08-13:
ah, missed that; my C++ is rusty anyway :)

>  I don't know better than that it should not enter config mode on a single(press) reset or after a save of the config, it should restart and nothing else.

after a config save it goes into normal mode, only after a single reset while in config mode it seems to go back to config mode...

pppedrillo, 2021-08-14:
> after a config save it goes into normal mode, only after a single reset while in config mode it seems to go back to config mode...

It can do this in case it couldn't connect to network specified in config (eg wrong SSID or password )

ErikdBr, 2021-08-14:
> > after a config save it goes into normal mode, only after a single reset while in config mode it seems to go back to config mode...
> 
> It can do this in case it couldn't connect to network specified in config (eg weak signal, wrong SSID or password etc)

As you can see in the successful session log where it connects to wifi, it also starts a HTTP server at 192...

pppedrillo, 2021-08-14:
> As you can see in the successful session log where it connects to wifi, it also starts a HTTP server at 192...

In his logs it doesn't connect, but rather created an access point. 
Then iSpindel connects to wifi successfully, it doesnot starts http server.

The reason it starts an access point (goes into config mode) -
**Double Reset detected**

ErikdBr, 2021-08-14:
> > As you can see in the successful session log where it connects to wifi, it also starts a HTTP server at 192...
> 
> In his logs it doesn't connect, but rather created an access point.
> Then iSpindel connects to wifi successfully, it doesnot start http server.
> 
> The reason it starts an access point (goes into config mode) -
> **Double Reset detected**

I was talking about the second log there it is also creating an access point while there is no reason to do so:
>**power-cycle or reset detected, config mode**

pppedrillo, 2021-08-15:
>> As you can see in the successful session log where it connects to wifi, it also starts a HTTP server at 192...

> I was talking about the second log there it is also creating an access point while there is no reason to do so:
> 
> > **power-cycle or reset detected, config mode**

Still can't see that it is connected to wifi, in both logs it just starts the AP.

m0n5t3r, 2021-08-15:
when it says `*WM: SET AP STA`, it means it has connected to wifi as well (it starts the builtin AP all the time, when it doesn't connect to wifi it only says `*WM: SET AP` like in the first example): https://github.com/universam1/iSpindel/blob/master/pio/lib/WiFiManagerKT/WiFiManagerKT.cpp#L335-L347