universam1/iSpindel #535: Fully resetting an iSpindel

gingerneil, 2021-09-01:
morning all...
When we use 'factory reset' the SSID is still shown in the config page (although crossed out when in config mode). I have tried to reflash the firmware, even tried to overwrite all the flash with zeros... but I cant get rid of it and fully reset the wemos. Is there a solution for this ?

pppedrillo, 2021-09-09:
https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html#persistent
**"By default, these settings are written to specific sectors of flash memory...."**

Read [here ](https://github.com/universam1/iSpindel/issues/507#issuecomment-862164046) how get rid of it. You need properly wipe all the flash.
[Here](https://github.com/universam1/iSpindel/issues/507#issuecomment-861723973) is a brief process desc.

thegreatgunbantoad, 2021-09-09:
Brewflasher will do a full wipe for you: https://github.com/thorrak/brewflasher/releases

closed 2021-11-16 by stale[bot]