universam1/iSpindel #545: Feature Request - MPU6500 Gyro

Bananamannn, 2021-10-07:
## I'm submitting a ...
  - [ ] bug report
  - [x] feature request
  - [ ] support request => Please use the [Discussion](https://github.com/universam1/iSpindel/discussions) board instead!

## Did you follow the general troubleshooting steps first:

- [x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x] Checked the logs for errors in the terminal console ?
- [x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Features
Support for MPU6500 gyro. I accidently bought a batch of those instead of MPU6050 units for 20 ISpindel units I'm building for my homebrew club (I know, spud move). So I'm pleading with the GitHub community to help out a fellow homebrewer trying to muddle his way through.

Or if anyone can point me in the right direction of something to read to get me started making the code changes myself that would be super helpful.

Bananamannn, 2021-10-08:
After some hours of reading and having an epiphany I've managed too get the standard code using the library for the MPU6050 working by commenting out line 68 in file MPU6050.cpp and replacing it with "true". I believe the error is occurring because the device ID of the MPU6500 is not 0x38. After looking at the datasheet it should be 0x70, but that doesn't work either. Any ideas??

```
bool MPU6050::testConnection() {
//    return getDeviceID() == 0x38; 
    return true;
}
```

stefschin, 2021-10-09:
In Datasheet says:
[image]
As I see in Schematics, by default, AD0 is connected low, so ID is b1101000 wich is 0x68
[image]

stefschin, 2021-10-10:
It's the only schematic I found as MPU 6500
https://www.smart-prototyping.com/MPU6500-6DOF-Sensor-Breakout-Board
I think it's same circuit for both ICs.
And in datasheet has same ID as MPU 6500

[image]

MPU9150 is a MPU6050 with integrated AK8975 magnetometer
MPU9250 is a MPU6500 with integrated AK8963 magnetometer

pppedrillo, 2021-10-11:
> 
> 
> > > I have tried "getDeviceID()" with both 0x68 & 0x69 and both give back an error
> > 
> > 
> > How many bits you was reading?
> 
> I'm sorry. I don't know how to test that :(

As I said  function getDeviceID() from the MPU6050 lib reads only 6 bits.
You need to read 7 bits to get any values grater than 0x3f as device ID

stefschin, 2021-10-14:
Sorry for delay but I was on a trip. I think we have made some mistakes about the code.
What I was talking about is Slave Address, but thinking that they are not the same for both devices. But reading datasheet both are the same: 0x68 for AD0 pin low and 0x69 for AD0 pin high. So nothing to worry until here.
You can see it on driver code: MPU6050.h line 60

#define MPU6050_ADDRESS_AD0_LOW     0x68 // address pin low (GND), default for InvenSense evaluation board
#define MPU6050_ADDRESS_AD0_HIGH    0x69 // address pin high (VCC)
#define MPU6050_DEFAULT_ADDRESS     MPU6050_ADDRESS_AD0_LOW

and has the purpose to communicate with different devices on same I2C bus.

In other side to check if device is correctly connected, you can read Register 117, WHO_AM_I, that returns the contents of the upper 6 bits of the MPU-60X0’s 7-bit I2C address, in this case 0x34. 
As you can see on MPU6050.cpp line 68

 bool MPU6050_Base::testConnection() {
     return getDeviceID() == 0x34;
 }

This function reads ID on this way:

uint8_t MPU6050_Base::getDeviceID() {
    I2Cdev::readBits(devAddr, MPU6050_RA_WHO_AM_I, MPU6050_WHO_AM_I_BIT, MPU6050_WHO_AM_I_LENGTH, buffer, I2Cdev::readTimeout, wireObj);
    return buffer[0];
}

Now, for MPU-6500, it has same register 117, WHO_AM_I but is different:

[image]

As we can see it's 0x70 and has 8 bit instead of 6 returned before. 
Anyway surely there are other differences between this two ICs that can make a bad communication or sensor readings. Normally I would search for other driver that's specific for MPU-6500 and compatible with the driver that uses iSpindel code. Or edit MPU-6050 driver to make it work with MPU-6500...
I can do it with more time but unfortunately both devices are not recommended for new designs, new Motion Sensor that I see for Arduino is LSM6DS3 but I didn't check for Arduino Library.
Hope it helps, cheers

closed 2021-12-26 by stale[bot]