universam1/iSpindel #557: Doc link do not exist: https://www.ispindel.de/docs/tools/TestServer/Readme.md

cmoulliard, 2021-11-28:
## Doc error 

The following link - https://www.ispindel.de/docs/tools/TestServer/Readme.md mentioned within the doc page - https://www.ispindel.de/docs/README_en.html#test-server under the section `Test server` --> simple server returns a `Error 404`

thegreatgunbantoad, 2021-12-20:
The correct link would now seem to be: https://github.com/universam1/iSpindel/blob/master/tools/TestServer/Readme.md
It looks to be an issue in the English version but not the German.

thegreatgunbantoad, 2021-12-20:
I think that: https://raw.githubusercontent.com/universam1/iSpindel/master/docs/README_en.md
needs to read: [simple test server](../tools/TestServer/Readme.md) on line124

thegreatgunbantoad, 2021-12-20:
Pull request created: https://github.com/universam1/iSpindel/pull/561

cmoulliard, 2021-12-22:
Link is working now

[image]

closed 2021-12-22 by cmoulliard