universam1/iSpindel #558: Is it possible to reduce the accuracy of "battery" and keep 2-3 digits after the decimal point?

jsbjx, 2021-12-05:
calling HTTP
{"name":"iSpindel000","ID":2774563,"angle":88.98952,"temperature":26.625,"temp_units":"C","battery":3.972889,"gravity":33.05824,"interval":1,"RSSI":-46}HTTPAPI: posting

"battery":3.972889
The voltage is 6 digits after the decimal point. Actually useless. Other software call data display is too long

vytux-com, 2021-12-05:
Why? You can always round or truncate on display

jsbjx, 2021-12-05:
The number of digits displayed in Craftbeerpi is too long. No way to set a decimal point

vytux-com, 2021-12-05:
Sounds like an issue with Craftbeerpi application

jsbjx, 2021-12-05:
Craftbeerpi received the data without processing it and displaying it directly. So if you send a signal, you can handle it well. The receiving end does not need to be processed

ErikdBr, 2021-12-05:
Did you measure the voltage of your fully charged battery, checked it against the displayed voltage in the Info page of the iSpindel configuration, and filled in the battery conversion factor in the configuration of the iSpindel?
Maybe that will help, if not, there is a ton of software out there that displays the voltage of the battery with 2 decimals so while in Craftbeerpi the receiving end may not need to be processed, maybe it can be processed?

jsbjx, 2021-12-05:
![3610237cb7aa524b18214ae8cc92f77](https://user-images.githubusercontent.com/32890201/144745515-4de10a31-68af-4d16-af1b-a5231ea9392d.png
[image]
[image]
)
The settings should be correct, the voltage data seen by the serial port is 6 digits after the decimal point
"battery":3.972889
Is it necessary for iSpindel to output data with 6 digits after the decimal point?

ErikdBr, 2021-12-05:
You left the default battery conversion factor of 191.8 so the displayed voltage of 3.93 is probably not correct. If you want the correct voltage you need to measure the voltage directly at the battery with a multimeter as described in the FAQ:
```
How can I calibrate the voltage display?
From firmware 3.0 on you can adjust the conversion factor. By default, factor 191.8 is used.

You can adjust by this formula:
(displayed voltage / measured voltage) * current factor (191.8) = new factor

The new factor should be between current factor more or less 20% (191.8*0.8 - 191.8*1.2).
```
That doesn't necessarily solve your problem but if you want less decimals you'll need to find someone who wants to change that in the iSpindel firmware or do it yourself by putting in a pull request. Or as said before process it at the receiving end.

thegreatgunbantoad, 2021-12-08:
The iSpindle just sends a floating point number for Tilt, Temp, Gravity and Battery.  Presumably CraftBeerPi handles formatting the others correctly, otherwise you'd have lots of digits on your SG too.  This sounds like a limitation of CraftBeerPi.  It really should be up to the program displaying the data to handle the formatting.

universam1, 2021-12-08:
iSpindel is not responsible for the frontend, it is a measuring device only. Trying to solve such UI issues at this point totally misses the point of separation of concerns

closed 2021-12-08 by universam1

avollkopf, 2021-12-20:
Although closed, just for documentation that this is a RTFM topic. It is described in my CraftbeerPi4 documentation that digits can be changed for sensor data: https://openbrewing.gitbook.io/craftbeerpi4_support/master/craftbeerpi-4-server/dashboard#item-menu  (->Sensor Data in the table of this chapter)

I added that feature back un June (https://github.com/avollkopf/craftbeerpi4-ui/commit/5e5c2e9922f647e0019c09cfb2b2384a719bcc68)

[image]