universam1/iSpindel #560: Wemos V4

porschepete, 2021-12-19:
Getting everything together I just realized I ordered V4 Wemos D1 mini. Looking at schematics it should work but I see a 470 resistor already on the board at the RST to D0 sleep connection. Do I need to omit the BAT 43 from the PCB and just bridge connection (using Jeffery but same for cherryphillip)? Thank you for any help.

thegreatgunbantoad, 2021-12-20:
I wouldn't be surprised if it works fine with the BAT43 as per the V3.  Give it a bash I guess and see how is goes.

ErikdBr, 2021-12-20:
I guess V3.1 is EOL. https://www.wemos.cc/en/latest/d1/d1_mini.html

porschepete, 2021-12-20:
@thegreatgunbantoad Sounds good, I’ll go ahead with the build and include the diode. Should hopefully have everything here after the holidays, will update this with my findings so anyone forward doesn’t run into issues.  

@ErikdBr Yep that’s what I assume, their Aliexpress store does not even have them for sale. Just knockoffs is all I can see available.

ErikdBr, 2021-12-20:
> @ErikdBr Yep that’s what I assume, their Aliexpress store does not even have them for sale. Just knockoffs is all I can see available.
Please don't buy the knockoffs they're usually sheit.

porschepete, 2021-12-20:
@ErikdBr Yep that’s what I’ve read, I have the official V4s D1 minis on the way. Just wanted to make sure that new resistor they put in doesn’t cause any issues.

thegreatgunbantoad, 2021-12-20:
> > @ErikdBr Yep that’s what I assume, their Aliexpress >store does not even have them for sale. Just >knockoffs is all I can see available.
> 
> Please don't buy the knockoffs they're usually sheit.

For a long time that's all you could get.  They tend to either work or not, they don't tend to fail down the line.  But still get the legit stuff if you can, the fakers shouldn't be given money.  When I first started you just couldn't source the real stuff, glad they now have a store now.

sym0nd0, 2021-12-25:
I've just ordered some of the V4 (as all the V3 Pros I ordered have been shite).

pppedrillo, 2021-12-27:
@porschepete With the 470 resistor you could not get a reliable waking up from  the deep sleep. Here are alot of discussion about this issue , you can search for issues like "iSpindel sends its data only once", "iSpindel not always sends data" etc

PharmaJap, 2022-01-25:
I don't see a need to remove the resistor, as long as the jumper remains open? Just bridge with a BAT43 at the through-hole, unless I'm missing something?

An interesting quirk of the v4 is the lack of an input diode to remove. If anyone has a sacrificial v4 they can remove the USB-C socket from, it'd be interesting to see the traces underneath to figure out a workaround.

porschepete, 2022-01-29:
Finally got everything in the mail and all soldered up. Happy to report that everything seems to be working correctly so far, flashed without issue (using Linux) and deep sleep seems to be working correctly. Only time will tell with next batch to be sure but I am going to close this issue for now. 

> An interesting quirk of the v4 is the lack of an input diode to remove. If anyone has a sacrificial v4 they can remove the USB-C socket from, it'd be interesting to see the traces underneath to figure out a workaround.

@PharmaJap Correct, there does not appear to be any input diode present looking and the board and schematics of the V4. I did not want to sacrifice a board quite yet, just make sure not to plug into the Wemos with a battery installed.

closed 2022-01-29 by porschepete