universam1/iSpindel #564: Discussion: What about using a ispindel github org  (e.g.: https://github.com/ispindel, ...) ?

cmoulliard, 2021-12-23:
## Discussion

What about using the a `ispindel` github org (e.g. https://github.com/ispindel) to host all the ispindel projects such as
- This one - https://github.com/universam1/iSpindel/i
- Site and documentation (jekyll, ...) - https://www.ispindel.de/
- Test server or clients (csharp, ...)
- Integration and routing (IoT collected data, MQTT, influxdb, grafana, ...) 
?

WDYT ? @universam1 @thegreatgunbantoad

sikkepitje, 2021-12-24:
I think this is a great idea

pppedrillo, 2021-12-27:
It is not an issue with a firmware and thus should go  into "Discussions", no?  
Same as another one, created by you  https://github.com/universam1/iSpindel/issues/562

cmoulliard, 2022-01-04:
Do you want that I close this issue and create a https://github.com/universam1/iSpindel/discussions ? @pppedrillo

closed 2022-03-17 by stale[bot]

cmoulliard, 2022-03-21:
Do you want that we create a https://github.com/universam1/iSpindel/discussions ? @pppedrillo @universam1