universam1/iSpindel #588: History of tilt entity in HA not shown as line graph

WesleyE, 2022-08-16:
## I'm submitting a ...
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please use the [Discussion](https://github.com/universam1/iSpindel/discussions) board instead!

## Did you follow the general troubleshooting steps first:

- [x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x] Checked the logs for errors in the terminal console ?
- [x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

The configuration (`unit_of_measurement`) for the tilt is not supplied by the HA MQTT integration, causing the history of the tilt entity to be shown as a state change instead of a line graph:
[image]

Any unrecognized `unit_of_measurement` like `°` will make HA show the line graph. Not sending it shows the data points as in the screenshot. See the docs [here](https://www.home-assistant.io/integrations/template/#unit_of_measurement).

## Your Environment
- Firmware Version used: 7.2.1
- Service backend you have configured the iSpindel to talk to: Home Assistant / MQTT

universam1, 2022-08-17:
It looks like an HA bug to me. Can you reason why this is an iSpindel issue instead?

WesleyE, 2022-08-17:
Its the way HA functions. Without a unit it treats the value as a state (open, opening, closed, closing) and does not show the line graph. When adding a unit, it is treated like a continuous unit that they are able to graph. 
There are special units (like degrees C) that they graph in a special way, anything that is not in their special list will just receive the normal line graph treatment.
Hope this helps!

universam1, 2022-08-17:
Thanks makes sense

vytux-com, 2022-08-17:
> Its the way HA functions. Without a unit it treats the value as a state (open, opening, closed, closing) and does not show the line graph. When adding a unit, it is treated like a continuous unit [...]

How are you defining the sensor in HA? 

Because you should be able to set the unit_of_mesurment in the config without it having to be passed from the source for example my definition of CPU temp for one of my servers.
[image]

WesleyE, 2022-08-18:
In this case its configured by the iSpindel itself using the HA MQTT Auto Discovery topics.

PR #589 (2022-08-17): Add unit_of_meas to tilt for HA MQTT integration
This adds the unit_of_meas to the tilt Home Assistant configuration topic to let HA show a line graph instead of the state history.

Fixes #588
commit e65ccdba (2022-08-17): Add unit_of_meas to tilt for HA MQTT integration (#589)

This adds the unit_of_meas to the tilt Home
Assistant configuration topic to let HA
show a line graph instead of the state history.

Fixes #588