universam1/iSpindel #600: Documentation webUI states /r while action is probably /reset

GElkayam, 2023-01-09:
https://github.com/universam1/iSpindel/blob/22951183beb5d70dbaa782aa87378bdc9019f0e7/pio/lib/WiFiManagerKT/WiFiManagerKT.cpp#L945

universam1, 2023-01-09:
Please describe the issue first

ErikdBr, 2023-01-14:
The issue is that when you click the /r link from the maintenance page you get this error:
[image]

GElkayam, 2023-01-16:
Hi @universam1 , 
@ErikdBr is right, when clicking the /r link, you get a 404.
seems that in this commit, /r was changed to /reset (calling handleReset), so just need to align.
https://github.com/universam1/iSpindel/commit/e46bed1c341345bd8f53f69c8ae9d4da7b683a98#diff-e99471864a9b9c55df31c591e65a440ea3e22f1869c13667f1eb8254ca492d72L148

small cosmetic issue.

ErikdBr, 2023-01-16:
> https://github.com/universam1/iSpindel/blob/22951183beb5d70dbaa782aa87378bdc9019f0e7/pio/lib/WiFiManagerKT/WiFiManagerKT.cpp#L945
And shouldn't this line be: `page += F("<tr><td><a href='/reset'>/r</a></td>");`

closed 2023-04-02 by stale[bot]

GElkayam, 2023-08-06:
Hey @universam1 , I see Stalebot closed this issue? care to reopen?

universam1, 2023-08-08:
fix released in 7.2.2