universam1/iSpindel #602: Wrong tilt value

Basbosbaas, 2023-01-11:
## I'm submitting a ...
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please use the [Discussion](https://github.com/universam1/iSpindel/discussions) board instead!

## Did you follow the general troubleshooting steps first:

- [x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x] Checked the logs for errors in the terminal console ?
- [x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

When I've callibrated the Ispindel on a flat surface its about 90 degrees tilt. But when I take it vertical it haves a tilt of about 45 degrees.

## Your Environment

- Firmware Version used: 7.2.1
- Browser Name and version: android

thegreatgunbantoad, 2023-01-13:
You may well have a duff sensor.  There are a lot of poor quality counterfeit items out there (Especially the Wemos boards).  I have had my fair share of rubbish tilt sensors.  I assume you have checked that it's the right way around and all that kind of thing.  A picture of it laid down during calibration might help us out.  This is unlikely to be a software bug.

ErikdBr, 2023-01-14:
Today I build a new iSpindel based on TheJeffrey 2.69 PCB and an original Wemos (Lolin) V3.1.0. I flashed it with the latest stable firmware: 7.2.1, did the calibration on a flat surface and these are my numbers, flat close to 90 degrees, vertical close (enough) to zero degrees in either direction (with the battery up or down), don't mind the gravity cause I didn't yet did the calibration in water so the default polynominal is still in there.
[image]
[image]

Basbosbaas, 2023-01-14:
Ok, I get it. 

Unfortunately it is a bad gyroscope. I will buy some more and test it with new ones.

Thanks for all the quick responses.

closed 2023-01-14 by Basbosbaas

Basbosbaas, 2023-04-13:
Just for you info: I've bought new gyroscopes and everything works fine!