universam1/iSpindel #604: Holder for the PCB card in the container

sonite, 2023-02-20:
## I'm submitting a ...
  - [x] feature request

## Did you follow the general troubleshooting steps first:
- [x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report
Constant recalibration issues after charging or improper handling (vibrations).

## Context
If the iSpindel is exposed to vibration/handling/pcb removal or moved during charging, current calibration might be offset. Require the user to redo all the calibrations. I want to mitigate this issue by a glued card holder in the bottom that will always provide a snug fit.

## Features
Will add convenience but cost to the product. Drop glue in the bottom of the petling, Push down a holder for the pcb card. Let it dry. Push down the pcb in the holder and calibrate. Now with this holder, the user can remove and insert the pcb card without changing the position. No more recalibration is needed.

Pros: The iSpindel will be resilliant to pcb removal/charging and vibrations.
Cons: Extra 0.01$for a designed/printed holder in the bottom

sonite, 2023-02-20:
Example:
[image]

ErikdBr, 2023-02-20:
To which design are you referring? Because in iSpindels build with the Jeffrey V2.69 PCB this will not fit.
[image]

sonite, 2023-02-21:
Very well defined question, thanks! I would then suggest that either version 3.0 Jeffery either is a bit shorter or that a "bulkier" design of the end-tube holder is considered. The above holder was only to illustrate and give an idea. Maybe another design of holder can be built that fit V2.69?

ErikdBr, 2023-02-21:
The Jeffrey V2.69 is already designed to fit tight in the tube. I build several and had no issues with calibration that need to be done again. You can even get the hole thing out of the tube, it will go back in the same as it got out.
But you can always ask the maker of the Jeffrey, you can find him here: [https://www.opensourcedistilling.com/the-jeffrey-2-69/](url)

Ventusfahrer, 2023-02-21:
I built 5 Spindles using Jeffry 2.69. There I had to sand the (3?) edges of the PCB, that it fits into the Petling. You can do that in a way, that the PCB has no play in the Petling. 

But you need to keep in mind, to smooth the sanded edges of the PCB. 

If not smoothing them, you will get scratches on the inner side of the Petling with a chance of slight leakage between the Petling and the lid.

sonite, 2023-02-22:
These are good comments. Some ppl buy these pills from others that have assembled them. As a user, I want to minimize the risk of recalibration just because I was unwary when charging it up. Also, a producer of the board might sand down too much and still sell the product, then the user will experience this as well. A small holder could mitigate this user issue.

Ventusfahrer, 2023-02-22:
I don't think, that this is possible with the Jeffrey PCB. There is no space left for such a centering.

Furthermore the PCB is located outside the axis, the Battery contacts the inner surface of the Petling. Together with the PCB contacting the surface too, you have an ideal 3 point mounting. At the bottom of the Petling, the PCB has a peak which needs to be sanded and at the opposite side I had to slightly to sand the two edges with the holes. When the lid is closed there is really no space left within the petling in any direction.

I can shake mine as strong as I want, there is no rattling to hear.

[image]

PCB is not centered in the Petling tube.

[image]

The battery and two edges have contact to the tube, the peak of the PCB to the bottom of the Petling and the two noses at the PCB-end have contact to the lid. But only after precise sanding. 

When using the your proposed centering you have exactly the same problem. Furthermore it will be very challenging to get the centering through the adjustment process in the tube and back out again, because it is relatively short. 

The Petlings I got are not exactly cylindric,  slightly conical to the bottom of the tube. The inner diameter of the petlings also differs slightly from one petling to the next and the are not exactly circular

thegreatgunbantoad, 2023-02-22:
I don't think I've had a need to remove the PCBs once they are in.  I just hot glued mine in, easy to cut that off if I need to get it out.  Only reasons I can think to pop it out are, battery dies, or an update crashes half way through and you need to reflash.

sonite, 2023-02-23:
Hot Glue it while when assemble it would also be a good "hack". Maybe not so customer friendly, but would work :)

thegreatgunbantoad, 2023-02-23:
I've made a few up for a FB group I'm with, but I'm not sure if you would be breaching anything selling as a business.  It is a question that has come up before.  I do know that in the US these would breech Tilt's copyright if sold.

sonite, 2023-02-23:
:) I'm not going to, I'm just talking for us noobs out there that accidentally moved on the card and find it to be a pain point. A pain point that I would like to mitigate

thegreatgunbantoad, 2023-02-24:
yeah I did run into that hence just gluing it in.  I assume that you know that you can do updates via wifi and those two pins at the end of the tube short to do a reset.  So there's no particular need to remove the PCB.  I replaced the rest pins with a magnetic reed switch and chucked wireless charger in just to prove that you can make one that you never need to open the tube, but it kept rolling of the charger so I just went with standard builds after that.  I use the cherry philip board it works very nicely.

sonite, 2023-02-24:
I made a 10 point calibration. Then I charged the card., removing the charging cable moved my card, and I had to redo all messy calibration ;)

thegreatgunbantoad, 2023-02-26:
I did similar first time around.  These days I never rely on the iSpindel for an accurate SG value to be fair, I just use it for the trend and to see when line flattens off.  I use floaty glass hydrometer readings before and after to get the ABV.

closed 2023-05-21 by stale[bot]