universam1/iSpindel #610: MQTT topic to force iSpindel in configuration (AP) mode?

pe1rmg, 2023-10-23:
## I'm submitting a ...
  - [ ] bug report
  - [X] feature request
  - [ ] support request

## Features
**Please note by far the quickest way to get a new feature is to file a Pull Request.**

I am using an MQTT setup with Home Assistant. I would love to have the option to request via MQTT the iSpindel to return to config AP mode? In essence just check if a certain topic (like "configrequest") is available with a value != 0. If it reads the topic as != 0 reset the iSpindel in config AP mode. Then during the writing of the config, also write this topic back to 0.

I am not familiar enough with PIO, and am unable to get the development environment setup. (It can not locate / download certain libs). Otherwise I would give it a try myself to implement.....

I posted this as well in the closed issue #96 and I read #210 , #255 , #422 too. Since what I request might be a lot less battery / processing intensive, it might be an option to implement this?

thegreatgunbantoad, 2024-03-17:
I think the device only sends out data it doesn't read anything back.  Also it only wakes up every 15 mins (or whatever you have it set to) so you'd have to wait 15 mins for it to pick it up if it could read back.  Maybe try attaching something like this to the reset pins: https://www.ebay.co.uk/itm/285455182911

thegreatgunbantoad, 2024-03-17:
I have also tried a reed switch and a magnet allowing you to reset by waving a magnet.  This along with a wireless charging pad then enables you to seal the unit and never need to open it.  One issue is that you can't really power it off (I did have a plan to have a magnetic stand and another read switch at the other end to over come that but the wireless charge pad took ages to charge so I abandoned the idea).  That said none of that is much help once it is in the fermenter.

pe1rmg, 2024-04-24:
@thegreatgunbantoad Sorry for my tardy reply.... I have already a solution to charge the iSpindel without requiring to opening it up. (It is not wireless actually...!) I have been running the iSpindel on the 18650 battery for months (Since early December) now, reporting every 15 minutes, and it is still going strong. So I am not too worried about the charging.
For me waiting for 15 minutes to get the iSpindel in "configuration mode" is not a problem, if that allows me to keep the unit sealed. 
What I understand of the MQTT protocol implementation is that it should be not too many lines of code to check if in a topic an entry exists, and act upon that existence.
I think I will setup a VM, and try again with PIO. Maybe that will work out :-)

pe1rmg, 2024-05-05:
@thegreatgunbantoad I am going to give it a go tomorrow.... Let's see if I am more successful this time! If I need help I will let you know.

closed 2025-05-06 by stale[bot]