universam1/iSpindel #614: Exception (28) on plain modules

lowPerformer, 2023-11-24:
## I'm submitting a ...
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please use the [Discussion](https://github.com/universam1/iSpindel/discussions) board instead!

## Did you follow the general troubleshooting steps first:

- [x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x] Checked the logs for errors in the terminal console ?
- [x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

Flashing firmware on a plain module (not installed on iSpindle PCB) throws an exception at boot.
This is due to the missing hardware, especially in the function:
```
void requestTemp()
{
  if (!DSreqTime)
  {
    DS18B20.requestTemperatures();
    DSreqTime = millis();
  }
}
```
DS18B20 never gets initialized when no sensor found (s.a. initDS18B20())

My workaround was:
```
void requestTemp()
{
  if (myData.OWpin == -1)
    return;
    
  if (!DSreqTime)
  {
    DS18B20.requestTemperatures();
    DSreqTime = millis();
  }
}
```

### Console Logs

```
FW 7.3.3
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
ERROR: failed to load json config

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 0
No devices found!
scanning for OW device on pin: 0
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Acc Test Connection ERROR!
offsets not available
Boot-Mode: Deep-Sleep Wake

Double Reset detected
...........
ERROR no Wifi credentials

going to Config Mode
Acc Test Connection ERROR!

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x402106dd epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffed60 end: 3fffffb0 offset: 0150
...
```

## Context
Wanted to flash several modules an configure them prior to mounting them on the PCBs. This is impossible with this bug.
This may also happen if the DS18B20 is malfunctioning in some way and is not found on boot-up!

ErikdBr, 2023-11-24:
It is not a bug, works as designed.

You can flash the module prior to assembling but to configure it you have to have it fully build.

universam1, 2023-11-24:
Closing as per design

closed 2023-11-24 by universam1