universam1/iSpindel #621: If gyro sensor is mounted inverted, the offset calibration does something that makes the unit not read less than 60 degrees when orientated vertically

Mark-Kiwi, 2024-06-30:
## I'm submitting a ...
  - [ ] bug report
  (hopefully this will also help others identify what is happening if they are having the same issue)
 
## Did you follow the general troubleshooting steps first:
Yes. then I hunted the internet high and low and only found one reference on a forum to a problem where the person could not get their iSpindel to read below 60 degrees when vertical. There was no solution posted

## Report

As my pelting tubes are smaller than what everyone else seems to be able to get, I had to build my own unit.
I found a sled on thingiverse that fits nicely and assembled my tube. (https://www.thingiverse.com/thing:2169567)

After installing the latest firmware (7.3.3) I verified the unit reads horizontal (approx 87 degrees) and vertical (near 0)
I then place the unit on a level surface and run the offset calibration. Then repower the unit.
Then when I check the horizontal and vertical (Horizontal is now 89.9 which is expected), the vertical reading wont go under 60 degrees so I cannot adjust the balance to make the unit float at 25 degrees in water.

I evetually figured out, that while most probably use trays or circuit boards where their gyro is orientated one way, the unit I have settles 180 degrees so the gyro board is inverted. Running the offset calibration while the gyro is inverted causes the issue.

### Console Logs
Didn't give me any clues

## Steps to Reproduce (for bugs)

1. load the latest firmware onto a unit or do a factory reset and repower the unit.
2. Confirm it can read expected values when the unit is orientated both horizontally and vertically. (this proves the gyro is working)
3. place the unit hoizontal so it's ready to do the offset calibration, but then rotate the unit so the gyro board is now inverted.
4. run the offset calibration and repower
5. check the angle readings when the unit is horizontal and then vertical. The vertical reading won't go below 60 degrees (should be near zero)
 
## Context

Building my own iSpindel using 3d printed sled https://www.thingiverse.com/thing:2169567 because I couldn't source any other sized pelting tubes.

My plan will be to connect to a node-red server running on my home network using mqtt 

## Solution suggestion

If possible, modify the code so the offset calibration can be carried out regardless of the orientation of the gyro unit,
or specify that the gyro board must be orientated face up when the ispindel is horizontal and at rest.

## Your Environment

- Firmware Version used: 7.3.3 (but did also try some earlier versions. (Same issue)
- Browser Name and version: Safari on iphone
- Service backend you have configured the iSpindel to talk to: - haven't proceeded beyond trying to do the offest calibration

Just wanted to say thanks and great work to the developers! Can't wait to get this unit working.

thegreatgunbantoad, 2024-06-30:
I can confirm that if you lay a typical build (Cherry Phillip board) in the correct position and rotate it through 180 deg around it's long axis (i.e. gyro facing down) that I see the same behaviour.  However, is there a reason you can't calibrate your unit in an orientation such that the led on the gyro would be skyward?

Mark-Kiwi, 2024-06-30:
The frame I used orientated the wemos in a way which allowed access to the
reset pin but also means the gyro goes one way, and the battery (being the
heaviest item is on top), naturally rotates the whole unit when it rests
and hence flips the electronics over. Now that I’ve identified this, I’m
going to modify the tray so that the wemos is installed the other way up,
and thus bringing the gyro to the top when the unit is at rest.

kaihuang201, 2025-03-08:
Wierd, I calibrate the my gyro led facing down it works just fine. Have you tried zeroing it?

closed 2025-06-27 by stale[bot]