universam1/iSpindel #626: İspindel wifi problem

Rasim06, 2024-11-15:
please help. ispindel program is installed successfully. but wifi does not turn on.

3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 116
No devices found!
scanning for OW device on pin: 121
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Acc Test Connection ERROR!
Boot-Mode: Deep-Sleep Wake
...........
ERROR no Wifi credentials

going to Config Mode
Acc Test Connection ERROR!

Exception (28):
epc1=0x4020e5fb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3ffffd60 end: 3fffffb0 offset: 01a0
3fffff00:  0031d260 3ffef1a0 3ffefa40 4020ef0a  
3fffff10:  00000000 00000000 3ffef7d0 40209174  
3fffff20:  0031d260 3ffef1a0 3fff2b44 40209aac  
3fffff30:  402466b4 3ffe9ca4 3ffef1a0 4021dca6  
3fffff40:  4024d070 3fff2b44 3ffef1a0 4021e166  
3fffff50:  60000600 3fff2b44 3ffef1a0 4024d07d  
3fffff60:  4024d0c2 3fffdab0 00000000 3fffdcb0  
3fffff70:  3ffef1b0 3fffdab0 00000000 00000004  
3fffff80:  40000f49 40000f49 3fffdab0 40000f49  
3fffff90:  40000e19 40001878 00000002 00000000  
3fffffa0:  3fffff10 aa55aa55 000000ec 4010538c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 116
No devices found!
scanning for OW device on pin: 121
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Acc Test Connection ERROR!
Boot-Mode: Exception
...........
ERROR no Wifi credentials

going to Config Mode
reboot RFCAL
s$

universam1, 2024-11-15:
The device only works fully assembled with sensors, no support for not finished device

Rasim06, 2024-11-15:
I am getting the same errors from my device built on PCB 4.

Rasim06, 2024-11-16:
problem solved :) Since I installed a 2-pin tilt sensor instead of a 3-pin on/off button, the sensors were not receiving power.