universam1/iSpindel #629: iSpindel looses Wifi connection after setup

Spezialfrosch, 2025-01-18:
Hi, 

i`m running a iSpindel on the latest Firmware 7.3.3
After loging into config mode i can setup Wifi SSID and PW, Service Type etc. 

When saving, the Spindel seems to connect to Wifi but looses connection immideatly. 
Wifi parameters are correct and checked with a local Hotspot with simple PW as well. iSpindel is listed as connected device for a while but disappears after a while. Therefore i can´t connect via local 192.168.4.1. The Spindel seems to accept the configuration as the devicename can be changed through configuration mode. Wifi led doesn´t blink after configuration.

Also tried older release and reinstalled 7.3.3 firmware.  

Any ideas?

Cheers, 
Spezi

ErikdBr, 2025-01-18:
This is normal behaviour. After saving the config it goes to normal operation and you should see results every xx seconds depending on what you filled in in the config. 
If you want to go to config mode again you have to repeat what you have to do to go to the config mode.

thegreatgunbantoad, 2025-01-19:
@Spezialfrosch as @ErikdBr notes, once you give it valid wifi credentials it won't serve its AP (Access portal) it will just send out data to the chosen server every xx seconds. You can only use 192.168.xx.xx when the LED is flashing blue, otherwise you are in normal mode.  Sleep, wake up send, sleep repeat.  Otherwise the battery just wouldn't last.

Spezialfrosch, 2025-01-19:
Hi guys,

My bad!
Somehow I assumed that I would still be able to find the Spindel. Additionally, I set the reporting cycle for testing to be much too infrequent.
Thanks for the quick help!

Spezi

closed 2025-01-19 by Spezialfrosch