universam1/iSpindel #633: DS18B20 readings 3~4°C higher than actual :(

nadir-dogan, 2025-02-16:
I have both original Dallas and Chinese DS18B20s. I used them on Leonardo and on ESP32-C6 unfortunately everytime results is 3~4 C higher than actual temperature. I'm working on this issue about one week. Not found any solution yet. If anyone have helpful knowledge I will be more happy. My subject is not to do any calibration. I'm looking for correct native readings. Thanks... #DS18B20

universam1, 2025-02-16:
The readings will stabilize once you go into a regular interval, not the config mode. Best is interval >15min. Reason is the self heating of the ESP8266 which accumulates the heat like a greenhouse in the container.

nadir-dogan, 2025-02-16:
Thank you for your reply @universam1 

[image]

[image]

I'm working on just prototyping phase, open frame open air. There is not any heater effect. I'm going to try 15 mins. intervals...

nadir-dogan, 2025-02-16:
It was closed 1~2 hours and I run it one time and the result is 25°C but in real the room is 22°C :(((
in the pictures one of them is Atmega32U4 other is ESP32-C6... 3 pcs. ds18b20 results are same...

closed 2025-05-06 by stale[bot]