universam1/iSpindel #634: Delay before first report

agentgonzo, 2025-03-02:
Is it possible to (optionally) make the ispindel do it's sleep/delay before sending the first message? 

It connects straight away as soon as I turn it on after charging. Given it's not floating in the beer at that point, it reports a wild number for the angle/gravity. Which then makes all my charts scale unnecessarily to take account of a huge or tiny SG value. 

It would be nice if I could turn it on, drop it in the fermenter, then have it send it's readings once it's settled

Thanks

ErikdBr, 2025-03-02:
I just delete the 1st reading from BrewSpy.

thegreatgunbantoad, 2025-05-23:
It kind of feels like the fix here needs to be in the server/cloud side, this is a flaw of those tools.  Brewfather does not have that problem, perhaps these other tools should be allowing deletion of data points or attaching the device after it is in place.