universam1/iSpindel #642: No tilt readings with new iSpindel firmware

MarkH62, 2025-05-05:
Apologies if this is a mistake on my part, but I've struggled with the problem now for several hours without success.
Initially today I had connection issue with my iSpindel. Depsite al efforts I was unable to connect it to our wifi, despite having used it sucessfully only 10 days ago. Ulitmately I refreshed the firmware to the latest version available and successfully connected the iSpindel to the wifi, however it is not returning any tilt angles other then 0.00.
I'd appreciate any help you can offer please.

lonneke94, 2025-05-22:
I am having same issue, should I load older firmware for fix?

thegreatgunbantoad, 2025-05-22:
Try doing a full reset, there was a change in the json file format and it might need to rebuild, that can get it a bit confused.

lonneke94, 2025-05-22:
I tried a factory reset from the maintenance menu and it still wont work... maybe Ill grab the latest firmware and try again

lonneke94, 2025-05-22:
Flashed firmware again with 7.3.3anf still no luck...

thegreatgunbantoad, 2025-05-23:
It your ESP board a genuine Wemos board?  The 3rd party ones can be flakey.  Also do double check your calculation.  Very wrong answers will be rejected by some tools.

lonneke94, 2025-05-23:
Ispindel pcb4.0 cherryphilip, i have reset to factory default 3 times and the info page says 0 and doesn't change if I move the ispindel around

thegreatgunbantoad, 2025-05-24:
Are you using plato?  Otherwise that calc looks off.
I'd be inclined to pop some brackets round the calc though just in case: (-0.00031*(tilt^2))+(0.557*tilt)-14.054.
I'd be surprised but that space between -0.00031 and the * might me upsetting the calc parser.

Think about it, it would be gravity showing zero not tilt so my thinking was a bit off there.

There was some trouble with the I'd of the gyros I'll have a look through the code and see what changed, but I thought the code had beem modded to be more accepting.

Also 7.1.4 is not the latest version, 7.3.3 is.  7.1.4 is from Octeber 2021.

wilbrod, 2025-05-24:
Did you build it yourself? Go over your soldering points.

MarkH62, 2025-05-25:
Thanks for the attention to this problem. I can confirm that my iSpindel was not 'self-assembled'. I'll take another look as soon as I can and post images as "lonneke94" has.

thegreatgunbantoad, 2025-05-26:
The improved accelerometer routine was added in at vesrion 7.2.0.  How are people flashing these, thorugh the AP or via Brewflasher?

lonneke94, 2025-05-29:
@thegreatgunbantoad I have reset to factory defaults via the config page and I have also pressed the reset button several times on the device to reset it and then I was grabbing .bin for 7.3.3 and then connecting through gui and adding the bin file for flashing.

lonneke94, 2025-05-29:
The biggest issue I'm having is that when I reset it and go to calibrate offset it switches from flashing blue to solid blue and then never finishes

ErikdBr, 2025-05-30:
The iSpindel doesn't have a reset button, it is a restart button when you press it one time and leads you to the config page when you press it several times short after eatchother. 

More information that could help:

https://www.google.com/search?q=solid+blue+light+on+ispindel

and

https://www.ispindel.de/docs/FAQ-en.html

MarkH62, 2025-06-10:
Has anyone been able to satisfactorily resolve this problem? The iSplindel is a paperweight sat on my desk when it should be in my next brew! I don't feel inclinded to replace it fearing a similar problem down the line.

thegreatgunbantoad, 2025-06-11:
@MarkH62 do you know how to hook it up and view the the diagnostic data?  Easiest on a Windows PC, but might be able to do something with an Adroid phone.

MarkH62, 2025-06-12:
@thegreatgunbantoad - no I don't, but would be able to with some guidance. Also, I bought this one ready assembled, so would anticipate it being a genuine part, but again, would welcome your help in confirming that, please.

thegreatgunbantoad, 2025-06-12:
@MarkH62 can you snap a picture of the board that has an Aerial track and a large metal housed device.  Opposite side of the board from the battery, tube opening end.  I suspect the issue is likely something to do with reprog or the Accelerometer.

Do you have a windows PC to try flashing via BrewFlasher (https://github.com/thorrak/brewflasher/releases or https://web.brewflasher.com/)?

thegreatgunbantoad, 2025-06-12:
@MarkH62 I'd also be inclined to see if the seller has experienced this issue before.

lonneke94, 2025-06-12:
[image]

lonneke94, 2025-06-12:
This is the one I have

MarkH62, 2025-06-15:
Hi @thegreatgunbantoad. Thanks for your help and patience!!
Here's the first image:
[image]
I've also flashed new software again, but still the problem remains. Here's the screen shot:
[image]
I've contacted the seller and the suggestion is to try flashing 6.5.1, but I don't know where to get that from?

thegreatgunbantoad, 2025-06-15:
@MarkH62 This does look like an issue with the accelerometer.  When I used to build these the accelerometers were on occasion unresponsive (Probabaly about 1 in 5).  But I never had one that had been working go on to fail.  There was an issue with some having a different ID code but I belive a fix went in for that.  Probably need to get a read out from the UART.  (Info here: https://opensourcedistilling.com/serial-monitor-log-for-ispindel/)

6.5.1 is here: https://github.com/universam1/iSpindel/releases/tag/6.5.1
You'll want the bin file.  And do a full reset via the AP after you reflash (At version 7.0.0 there was a filesystem change so any reflash that crosses this need a full reset via the AP)

ErikdBr, 2025-06-15:
> [@MarkH62](https://github.com/MarkH62) This does look like an issue with the accelerometer. When I used to build these the accelerometers were on occasion unresponsive (Probabaly about 1 in 5). But [...]

There is also an issue with the 4.0 version of the cherryphilip boards where the temp sensor solder side can scratch of the insulation of the battery causing a short circuit causing al kind of problems.

thegreatgunbantoad, 2025-06-15:
I'll put some info together about Installing Git, VS Code and Platform IO, Pulling the project and clean and build tomrrow.  But I'm only going to do that for Windows, Mac likely won't work as it just doesn't play well with others.  Linux guys I assume you were just born with **more** knowledge.

MarkH62, 2025-06-18:
Hi gents. I've now flashed with 6.5.1 but have this displaying now:

[image]

How do I do a full reset, as I don't think this will have 'happened'(!), and also, shouldn't this be done before I re-flash 6.5.1?

Kind regards, Mark

ErikdBr, 2025-06-18:
What should I do if my iSpindel isn't working properly after flashing it with the firmware?

For unclear reasons sometimes it looks like flashing of the iSpindel went well while it actually didn’t which could lead to unpredictable behavior. This behavior could be but is not limited to:

the blue LED of the wemos stays on permanently
parts of configuration is not saved, e.g. offset tilt
The solution to this can be initializing the memory of the wemos before flashing it with the iSpindel firmware.(as if you were formatting a new harddrive before you can use it)

Here is how to do that using NODEMUCU FIRMWARE PROGRAMMER

To initialize:

In the Config Tab

		1st column		2nd column
1st Line:	INTERNAL://NODEMCU	0x00000
2nd Line:	INTERNAL://BLANK	0x01000
In the Advanced Tab Set the Baudrate to 115200

In the Operation Tab Click Flash(E) and wait till the flashing is done

To flash with the iSpindel firmware:

In the Config Tab

		1st column		2nd column
1st Line: 	C:\iSpindel-6.0.6.bin 	0x00000
Where “C:\iSpindel-6.0.6.bin” is to be replaced by the path to your firmware file Make sure all the other lines in the Config tab are cleared and only the checkbox in front of your firmware path is checked.

In the Advanced Tab Set the Baudrate to 115200

In the Operation Tab Click Flash(E) and wait till the flashing is done

It now should work properly.

Source: https://www.homebrewtalk.com/forum/threads/ispindle-diy-electronic-hydrometer.598187/page-19#post-8366882

MarkH62, 2025-06-18:
Thanks for your time, Eric, however my problem has not been resolved ...

[image]

Any more suggestions would be welcomed.

In anticipation, Mark

thegreatgunbantoad, 2025-06-19:
This post is just going to describe what you need to install to use platform IO to talk to the iSpindel

So first things first
Windows support only, I don’t know Mac and if you chose Linux, you shouldn’t need help.
Install VS code (I tend to use the inser version)
Git
Python via cmd
When requested do NOT use VIM use VS Code:
 
[image]

Prove the basics are there
In VS code under Terminal in the menu bar open a ‘New Terminal’ type git –version
That’s should display git version <some version number> if not, stop and try again.
git needs to know who you are
in terminal type git 
git config --global user.name “Your user name”
git config --global user.email [youremail@domain.com](mailto:youremail@domain.com)

Now cd into a sensible directory and pull the project
cd C:\
git clone https://github.com/universam1/iSpindel.git
 
[image]

Now in VS Code install the platformIO Extension
 
[image]

Attach the usb and run windows update
In device manager Ports, should look like this:
 
[image]

May need to do Action->Scan for hardware changes

You'll also need python, you can grap this from the winows store by just typing 'python' into a windows command prompt and following where it sends you.

I'll do a second post about driving platform IO later.

thegreatgunbantoad, 2025-06-20:
Once you have all that set up, open VS Code and Select File->Open Folder... and navigate to C:\iSpindel and hit Select Folder.

Wait a short while for PIO to recognise the folder is a porject and set things up.

After this select the PIO icon 

[image]

Under d1_mini_git->Platform you should find Erase Flash

[image]
Select that to Erase, then under Defaut-General select Upload All to build the project and upload it.

thegreatgunbantoad, 2025-06-20:
@MarkH62 @lonneke94 Just at-ing you as I've put some instructions for a more involved wipe and reflash.  Assumption is that you're using windows.

MarkH62, 2025-06-22:
Thanks again. I am using Windows, buit this is probably at my limit, but I have a mate who I'll ask for assistance at the next opportunity and get back to you.

thegreatgunbantoad, 2025-11-23:
@loreStefani if you can get the diagnostics running that would help.  Otherwise look at replacing the accelerometer and after that the Wemos.